realsenseai/librealsense #429: Updating of firmware for RealSense R200 Camera

fw-timku, 2017-02-02:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | 16.04 LTS  |
| Kernel Version (Linux Only)  |   |

Hi all, I am currently using the Realsense R200 camera with firmware version of 1.0.71.10. I would like to know how do I go about updating the camera firmware to 1.0.72.10. What is the difference between the two firmware?

zubair123khan, 2017-03-09:
i am getting a warning :- 

[ WARN] [1489075664.860111150]: /camera/driver - Detected unvalidated firmware:
				- 2391012929's current camera firmware is 1.0.72.10, Validated camera firmware is 1.0.72.06
 

how to correct this.... now there is a problem if the camera firmware is up to date... strange !!

dorodnic, 2017-03-14:
What tools are you using to update R200 firmware? Did you try to update using the latest versions of [Digital Camera Manager](https://downloadcenter.intel.com/download/25044/Intel-RealSense-Depth-Camera-Manager)?
In general, librealsense is agnostic to specific firmware versions and there is no special treatment of the different R200 firmware versions.

molinadavid, 2017-03-24:
I have the same problem, I just updated the firmware on my camera using the official SDK and got the same message.

mdhorn, 2017-03-24:
@zubair123khan @molinadavid The WARNING you are seeing is from the realsense_camera ROS package, not librealsense.
It is only a Warning message denoting that the version of FW on hour camera is different from the version used to validated that release of realsense_camera ROS package. 
This is used as an indicator to our QA team to update the version of FW used for validation.

In general, the warning can be ignored; it is a warning, not an error.

closed 2017-04-03 by dorodnic