realsenseai/librealsense #431: No camera detected error after apt-get install

2Reeko, 2017-02-06:
| Required Info  |   |
|---|---|
| Camera Model | R200| 
| Firmware Version | 1.0.72.10  | 
| Operating System & Version |  Ubuntu 14.04.5 LTS |
| Kernel Version (Linux Only)  | 4.4.0-62-generic  |
| Build System |  Makefile | 

I just got a new machine and wanted to run R200 cameras. I have installed librealsense from source on an old machine before but wanted to try apt-get on this new machine hoping it would be easier.

What I did:
1. Before using apt-get to install librealsense I enabled the kernel sources via the script. 
2. `Sudo apt-get install ros-indigo-librealsense` threw no errors just a warning ` failed to verify signature on ./linux-lts-xenial_4.4.0-62.83~14.04.1.dsc`.
Terminal output:
```
sudo apt-get install ros-indigo-librealsense 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  ros-indigo-librealsense
0 to upgrade, 1 to newly install, 0 to remove and 0 not to upgrade.
Need to get 0 B/306 kB of archives.
After this operation, 1,218 kB of additional disk space will be used.
Selecting previously unselected package ros-indigo-librealsense.
(Reading database ... 302752 files and directories currently installed.)
Preparing to unpack .../ros-indigo-librealsense_1.12.0-0trusty-20161223-200153-0800_amd64.deb ...
Unpacking ros-indigo-librealsense (1.12.0-0trusty-20161223-200153-0800) ...
Setting up ros-indigo-librealsense (1.12.0-0trusty-20161223-200153-0800) ...

Creating symlink /var/lib/dkms/uvcvideo/1.1.1-3-realsense/source ->
                 /usr/src/uvcvideo-1.1.1-3-realsense

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Running the pre_build script:

dkms: Pre-build script started...

Downloading kernel sources...
Reading package lists...
Building dependency tree...
Reading state information...
Picking 'linux-lts-xenial' as source package instead of 'linux-image-4.4.0-62-generic'
NOTICE: 'linux-lts-xenial' packaging is maintained in the 'Git' version control system at:
git://git.launchpad.net/~ubuntu-kernel/ubuntu/+source/linux/+git/trusty -b lts-backport-xenial
Need to get 149 MB of source archives.
Get:1 http://au.archive.ubuntu.com/ubuntu/ trusty-updates/main linux-lts-xenial 4.4.0-62.83~14.04.1 (dsc) [7,369 B]
Get:2 http://au.archive.ubuntu.com/ubuntu/ trusty-updates/main linux-lts-xenial 4.4.0-62.83~14.04.1 (tar) [133 MB]
Get:3 http://au.archive.ubuntu.com/ubuntu/ trusty-updates/main linux-lts-xenial 4.4.0-62.83~14.04.1 (diff) [16.2 MB]
gpgv: Signature made Thu 19 Jan 2017 04:51:48 AEDT using RSA key ID FDCE24FC
gpgv: Can't check signature: public key not found
dpkg-source: warning: failed to verify signature on ./linux-lts-xenial_4.4.0-62.83~14.04.1.dsc
dpkg-source: info: extracting linux-lts-xenial in linux-lts-xenial-4.4.0
dpkg-source: info: unpacking linux-lts-xenial_4.4.0.orig.tar.gz
dpkg-source: info: applying linux-lts-xenial_4.4.0-62.83~14.04.1.diff.gz
dpkg-source: info: upstream files that have been modified: 
Fetched 149 MB in 16s (8,871 kB/s)
INFO: Only Intel RealSense(TM) R200 camera is currently supported.
Patching uvcvideo sources...
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file include/uapi/linux/videodev2.h
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
[... 151 lines cut ...]
process[camera/driver-2]: started with pid [17539]
[ INFO] [1486357054.250060939]: Initializing nodelet with 8 worker threads.
[ERROR] [1486357054.295632287]: /camera/driver - No cameras detected!
[ INFO] [1486357054.295671756]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1486357059.300296372]: /camera/driver - No cameras detected!
[ INFO] [1486357059.300419524]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1486357064.305109131]: /camera/driver - No cameras detected!
[ INFO] [1486357064.305248907]: /camera/driver - Sleeping 5 seconds then retrying to connect
[ERROR] [1486357069.309877511]: /camera/driver - No cameras detected!
[ INFO] [1486357069.310012615]: /camera/driver - Sleeping 5 seconds then retrying to connect
^C[camera/driver-2] killing on exit
[camera/nodelet_manager-1] killing on exit
[camera/nodelet_manager-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
```

What I've tried:
1. `lsusb` doesn't show a realsense camera connected.
2. `ls /dev` doesn't show video 1/2/3
3. `dmesg` after plugging a camera in shows no messages
4. `dmesg` after `sudo modprobe uvcvideo` also shows no messages

Seems like the cameras aren't recognised. Is there a step in my install that I missed?

dorodnic, 2017-02-06:
Hmm.. The fact that the camera does not appear in `lsusb` list and there is nothing in the `dmesg` indicate some sort of a hardware issue. Couple of questions that might give us more directions: are you running inside a VM? Is the camera connected to a USB3 (super-speed) port? Is the camera recognized on some another platform? Are you using the short USB cable provided with the device? Are you using a USB Hub for the camera? I know its a lot of questions, but these are based on common problems.

2Reeko, 2017-02-06:
Answers:
Are you running inside a VM? No, vanilla ubuntu machine
Is the camera connected to a USB3 (super-speed) port? Yes
Is the camera recognized on some another platform? Yes, 2 other machines with source builds can run the (4) cameras
Are you using the short USB cable provided with the device? Yes
Are you using a USB Hub for the camera? No, directly connected to the motherboard usb3.0 port

Extra info:
1. If I build from source then `dmesg` says `sudo modprobe uvcvideo` was successful
2. I am using the kabylake i7-7700k cpu on asus prime z270-ar motherboard. This seems to be the only notable difference between machines.
3. source builds also result in the same error. (librealsense v0.9.2 && realsense camera v1.4.0)

2Reeko, 2017-03-05:
Just to update and close.

Added new USB 3.0 hubs via pci-e ports:
-The apt-get install still doesn't work.
-Source builds now work fine. (librealsense v0.9.2 && realsense camera v1.4.0)

closed 2017-03-05 by 2Reeko

nr-parikh, 2017-05-23:
Hi 2Reeko,

Can you please elaborate on how did you solve the above mentioned solution? I am facing same issue. It would be a great help if you can elaborate on above mentioned solution!

Thanks in advance!

2Reeko, 2017-05-24:
Hi,

I gave up on the apt-get install because it was a pain.

If you just want r200 cameras then it is super easy to install from source and that is what I've done multiple times. Remember that you don't need to patch the kernel if you are up to date. Just follow the install instructions and skip the patch/backend sections then modprobe. 

If this doesn't do it for you then I suggest opening a new issue with all the relevant info. It is hard for me to help without knowledge of your basic set up.

Hope this helped

nr-parikh, 2017-05-24:
@2Reeko I was able to do solve the previous problem but now it shows `Input/output error` which is a uvcvideo problem. I tried to do `sudo modprobe uvcvideo`, it shows no error but at the same time, it also doesn't solve the problem. Here is what my setup is and what I did:
* Single chip computer UP board and single Realsense R200 camera
* Installed both __librealsense__ and __realsense_camera__ packages from the source
* Since the kernel was an older version I upgraded the kernel by running a script `install-dependencies.sh`
* I also patched connectivity workaround
* I patched __uvcvideo__ by running the script `patch-uvcvideo-4.4.sh`. It ran successfully
* As a final step, I did `sudo modprobe uvcvideo`

After all this I was able to solve camera detection problem but now I cannot start the camera stream.
Have you ever faced this problem?

2Reeko, 2017-05-25:
@nr-parikh 
If you are on older kernels then getting the cameras to work is a real mission. Also (I'm not sure if this is still relevant) there were extra steps if you were trying to install on arm processors, so that might be something to look into depending on which board your using. 

I've used this method (You'll have to manually delete any librealsense.so or other files that were made during the apt-get install as they are often left over even after purging):

1. Download and build librealsense first
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
git checkout -b b0.9.2 v0.9.2

2. Download and build realsense_camera second
git clone https://github.com/intel-ros/realsense
cd realsense
git checkout 1.4.0

These versions seem to be the most stable for using the r200. You may run into an error during the patch install and even worse is that it might not give you an error on screen for it. I used a few different closed issues to fix these problems. I can't remember the titles for them but one had to do with the config path and the other was to do with the url for cloning the ubuntu repo.

nr-parikh, 2017-05-25:
@2Reeko 
Apparently, it seems like the issue was with the cable being used. Interestingly both cable came in the both with two different cameras but one has USB 3.0 port and the other one has mini USB 3.0. The one with the USB 3.0 didn't work but the other one works perfectly fine. My best guess is that I was getting an error because of the voltage drop in the converter(USB3.0 to mini USB3.0) provided by Intel. This drop of voltage may have resulted in an insufficient power supply to the camera and giving me the error.

I will check by resetting everything and then trying again with the same condition to duplicated the error and try to validate it. 

Thank you for the support!!

furushchev, 2019-04-26:
Hello,
I recently saw the same situation on my environment with Realsense D435 and other cameras.
After I installed `librealsense2` stuffs, no video device is found as `/dev/videoX`.
Grasping at straws, I read up `dmesg` and found my computer is booted up with secure boot.
I disabled secure boot and rebooted and now found all video devices and realsense.
Maybe it is too late but hope it will help. :+1: