realsenseai/librealsense #432: rs_intrinsics coeffs[5] structure question

hlc111, 2017-02-07:
| Required Info  | rs_intrinsics coeffs[5] structure question  |
| Camera Model | R200| 
| Firmware Version |   | 
| Operating System & Version | 14.04   |
Hi,

About distortion coefficients  in rs_intrinsics we have 5 float data.
Could you kindly let us know which is radial distortion coefficients  (k1,k2,k3) and which is  tangential distortion coefficients (p1,p2).

Thanks!

ddiakopoulos, 2017-02-07:
As shown in the code here, https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense/rsutil.h, `coeffs[0`], `coeffs[1]` and` coeffs[4]` are `k1`, `k2`, and `k3` and and `coeffs[2] `and `coeffs[3]` are `p1`, and `p2`.

hlc111, 2017-02-07:
Thank you very much!

closed 2017-02-07 by hlc111