realsenseai/librealsense #436: Mac OS X make fails...

drhalftone, 2017-02-11:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Mac OS X 10.12.3 |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 

I'm able to properly configure make without issue, but when I run the make command I get the following error:

DrHalftone:build dllau$ make
Scanning dependencies of target realsense_log_utils
[  1%] Building CXX object sdk/src/utilities/logger/log_utils/CMakeFiles/realsense_log_utils.dir/log_utils.cpp.o
Linking CXX shared library librealsense_log_utils.dylib
ld: warning: -pie being ignored. It is only used when linking a main executable
[  1%] Built target realsense_log_utils
Scanning dependencies of target realsense_image
[  3%] Building CXX object sdk/src/core/image/CMakeFiles/realsense_image.dir/image_base.cpp.o
[  4%] Building CXX object sdk/src/core/image/CMakeFiles/realsense_image.dir/custom_image.cpp.o
[  6%] Building CXX object sdk/src/core/image/CMakeFiles/realsense_image.dir/image_conversion_util.cpp.o
/Users/dllau/SourceTree/RealSense/sdk/src/core/image/image_conversion_util.cpp:7:10: fatal error: 'opencv/cv.h' file not found
#include "opencv/cv.h"
         ^
1 error generated.
make[2]: *** [sdk/src/core/image/CMakeFiles/realsense_image.dir/image_conversion_util.cpp.o] Error 1
make[1]: *** [sdk/src/core/image/CMakeFiles/realsense_image.dir/all] Error 2
make: *** [all] Error 2
DrHalftone:build dllau$ brew search realsense
No formula found for "realsense".
==> Searching pull requests...

I just installed Opencv 3.2.0, after removing all prior versions.

dorodnic, 2017-02-12:
From the compilation errors, it seems you are trying to compile the [Linux SDK](https://github.com/IntelRealSense/realsense_sdk) on Mac. (The line `#include "opencv/cv.h"` is not part of librealsense or its dependencies).

drhalftone, 2017-02-12:
So how do I compile the Mac SDK?  I really just want basic access to the sensor for reading the color, depth, and ir channels.  I’m not looking for any tracking or human computer interfacing features.

dorodnic, 2017-02-13:
The Linux SDK is a layer on top of librealsense providing advanced functionality like recording and integration with Intel computer vision middle-ware. To the best of my knowledge there is no Mac version. librealsense does seem to offer what you need (reading color, depth, IR) and it works on Mac. Just follow to installation [instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md) and you should be fine. There seem to be a small [issue](https://github.com/IntelRealSense/librealsense/issues/401) with newer versions of OSX but you will find [the following pull-request](https://github.com/IntelRealSense/librealsense/pull/408) useful. (Intel does not officially support RealSense on OSX at the moment, so we are relying on community involvement to help maintain librealsense on OSX)

drhalftone, 2017-02-20:
These instruction for Mac only describe installed dependencies, nothing about getting the basic RealSense libraries for interfacing with the camera.  I'm trying to compile and install, "librealsense.dylib".  How do I do that?

ddiakopoulos, 2017-02-20:
Hi @drhalftone -- please look in this directory: https://github.com/IntelRealSense/librealsense/tree/master/librealsense.xc 

If you load the Workspace up in XCode, it will compile the core library and 10+ examples

drhalftone, 2017-02-20:
Yes, I did that and had a lot of problems with not finding the opencv and opencv2 header directories.  So I manually changed all the "opencv/cv.h" and similar header includes and put absolute paths in place.  I tried setting the header directories in Xcode, but to no avail.  So yes, I am now able to compile.  Thanks for the suggestion.

closed 2017-02-20 by drhalftone