realsenseai/librealsense #437: VIDIOC_DQBUF error 19 .. for Realsense_R200 on Upboard

szx0112, 2017-02-13:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | ubuntu 14.04  |
| Kernel Version (Linux Only)  | 4.4.0-62-generic  |
| Build System |  | 

I know this problem has been discussed before at ( [](https://communities.intel.com/thread/110133) and [](https://communities.intel.com/thread/110133) ). It concluded that the problem is related to the power supply of r200 and can be resolved by connecting the micro B to micro B port directly with USB 3.0 cable from camera to Upboard.

However, the problem I am facing is that the camera sometimes can run smoothly when calling the launch files in realsense_camera ROS package for a random number of minutes (sometime it can last over 20 minutes, but most of time only works for a couple of seconds). Then the errors VIDIOC_DQBUF error 19 showed up, which represents the camera is disconnected (it also can be identified with the disappear of the red spot on r200).

Anyone with the idea of this problem. I believe everything are set up right based on the documentation. Thanks.

szx0112, 2017-02-13:
Kind of find the problem but not sure how to resolve it, the camera will automatically disconnect and reconnect during operation. After the reconnection, the usb port will assign a new location for it that probably mis-matched with the original settings.

dorodnic, 2017-03-09:
VIDIOC_DQBUF resulted by camera disconnections are common side effect of insufficient power, and can be worked-around by connecting the camera through a powered USB3 hub. Unfortunately, the software has no control over this effect.

closed 2017-03-09 by dorodnic

szx0112, 2017-03-09:
Thanks for reply. I use the USB3 hub on Upboard. So does it mean the usb3 hub on Upboard is also under powered? Or there is any way to increase the power? Thanks.

dorodnic, 2017-03-14:
When connecting the camera through external USB3 hub, with the hub itself plugged to the wall, we see very stable results (hundreds of hours of streaming). When the camera and the Upboard share a single power source, we do encounter occasional disconnects. These are the data points I can provide regarding R200 performance on the Upboard. However, the librealsense library has no control over it.

renno-bih, 2018-11-06:
> When connecting the camera through external USB3 hub, with the hub itself plugged to the wall, we see very stable results (hundreds of hours of streaming). When the camera and the Upboard share a [...]

What USB 3.0 HUB are you using? When connecting R200 using cheap USB  3.0 HUB (powered) it cannot be detected.