realsenseai/librealsense #439: How to Identify Camera from USB?

nwestbury, 2017-02-14:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |  3.10.10.0 & 3.17.0.0 | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |   4.4.0-62 |
| Build System |  Makefile | 

This issue might be impossible to solve. I have two SR300 cameras, one with 3.10.10.0 firmware and another with 3.17.0.0. I tried plug each camera into two separate ports. The one with the lower firmware reports two separate ports (7-1 and 7-2). Whereas the one with the higher firmware reports the same for both (7-1 and 7-1). I get this ports by calling `rs_get_device_usb_port_id` (which internally calls libusb). I understand [ports are not guaranteed to be unique](http://libusb.sourceforge.net/api-1.0/group__dev.html#ga14879a0ea7daccdcddb68852d86c00c4) however this is on the same hardware with only slightly different cameras.

In addition, running `v4l2-ctl --list-devices` gives two different formats (this is on hub-port 7-2): 

3.10.10.0
Intel(R) RealSense(TM) Camera S (**usb-0000:00:10.0-2**):
	/dev/video0
	/dev/video1

 3.17.0.0
Intel(R) RealSense(TM) Camera S (**usb-0000:00:10.0-2.1**):
	/dev/video0
	/dev/video1

Any idea why this behaviour is different and how to resolve it?

dorodnic, 2017-04-03:
@nwestbury, perhaps I didn't understand the issue correctly. Different models of SR300 are reporting `port_id` differently because some manufacturers choose to include internal USB hub inside the camera case, and some don't. The goal of `get_port_id` is to distinguish between devices based on physical location and not concrete serial numbers. On-board USB-hubs certainly complicate things a bit, but we can't know from code if the USB-hub is part of the device or not (hope this makes sense). 
Could you please clarify your question, perhaps add the output from `lsusb -t` for both devices?

nwestbury, 2017-04-03:
@dorodnic Yes, you are correct it was just USB hub inside the camera case.

closed 2017-04-03 by nwestbury