realsenseai/librealsense #440: rs.error: Couldn't query adapter board / motion module FW version!

DonnaWuDongxia, 2017-02-15:
| Required Info  |   |
|---|---|
| Camera Model | R200, F200 or SR300 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile |

DonnaWuDongxia, 2017-02-16:
@mattcurfman have you ever encountered the similar issue? I run slam samples in MW, it suffers the same problem.

DonnaWuDongxia, 2017-02-16:
I have no idea if it is a hardware issue or a firmware issue.

sonyccd, 2017-04-20:
Having same problem

squatsandscience, 2017-05-27:
Also having the same problem. Any clue @dorodnic ?

haoyunfeix, 2017-06-07:
I met the same problem on Joule board with Ubuntu16.04.
```
~/weekly/0.10.0/node-realsense/src/slam/test$ cpp-motion-module
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense ZR300
    Serial number: 3511804691
    Firmware version: 2.0.71.28
This device does not support motion tracking!
```

jlgarcia75, 2017-10-03:
Hello,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75

jixiedaima, 2017-12-26:
I get the same problem,do you resolve it?

tony497, 2018-03-20:
Same problem with R200

```
Precision-7520:/usr/local/bin$ ./rs-depth 
There are 1 connected RealSense devices.

Using device 0, an Platform Camera
    Serial number: 1-11-4
    Firmware version: (null)

rs_error was raised when calling rs2_get_device_info(dev:0x1b45880, info:Firmware Version):
    info Firmware Version not supported by the device!

```
Anybody has an idea ?