realsenseai/librealsense #444: Uvcvideo patching issue

Ryanliyikun, 2017-02-21:
| Required Info  |   |
|---|---|
| Camera Model |  SR300 | 
| Operating System & Version | Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.8.0-36-generic  |
| Build System |  VS2013, XCode or Makefile | 

I installed Ubuntu on my mac as a dual system. When I run the script ./scripts/patch-realsense-ubuntu-xenial.sh I got the error : 
modprobe: ERROR: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version

dmesg : [ 3233.439976] uvcvideo: Unknown symbol media_entity_create_link (err 0)
[ 3233.440057] uvcvideo: Unknown symbol media_entity_init (err 0)
[ 3233.440100] uvcvideo: Unknown symbol media_entity_cleanup (err 0)

I got the same error when I tried to install it on a windows/Ubuntu dual system machine. I solved it by disable secure boost with command below:
sudo apt install mokutil
sudo mokutil --disable-validation

It worked. However, it does not work on my mac. When I tried to run the commands. It says:
 This system does't support Secure Boot
What should I do to solve this ?
Thank you!

ev-mp, 2017-02-22:
Hi, you've got Ubuntu LTS with kernel 4.8, probably '**_16.04.2_**' that was released this month.
The patch, however,  support LTS kernels 4.4.0-xx only (4.4.0-63 the latest) .

Support for kernel 4.8 is on the TODO list, but at this point you should either downgrade Ubuntu to kernel 4.4, or modify the patches on your own to make this work.

hiptechboy, 2017-02-28:
Is there an ETA for kernel 4.8 support?

jamessergeant, 2017-03-09:
I'm also interested in ETA for 4.8 support. Will downgrade kernel until then.

mindThomas, 2017-03-20:
We have tried downgrading to 4.4 on an Intel Aero board but do unfortunately not have any success with enumerating the camera.
Looking in dmesg shows that the USB 3.0 port is started up and detected but fails afterwards why the port become "deregistered".
```
[    7.013643] xhci-hcd xhci-hcd.3.auto: xHCI Host Controller
[    7.013662] xhci-hcd xhci-hcd.3.auto: new USB bus registered, assigned bus number 3
[    7.048224] xhci-hcd xhci-hcd.3.auto: Host not halted after 16000 microseconds.
[    7.048233] xhci-hcd xhci-hcd.3.auto: can't setup: -110
[    7.048315] xhci-hcd xhci-hcd.3.auto: USB bus 3 deregistered
[    7.048349] xhci-hcd: probe of xhci-hcd.3.auto failed with error -110
```
Any idea?

eric-schleicher, 2017-04-19:
Requesting update! 16.04.2 is the mainline LTS released version.  Running 14.04 doesn't work for many robotics applications.  

We mark realsense as non-op/no-use until this is fixed.

tdnet12434, 2017-04-22:
Someone found the way to fix this issue yet?

Thanh-Binh, 2017-05-05:
I have the same problem with uvcvideo for Ubuntu 16.04lts, kernel 4.7. Is the proposal "disable secure boost" the final official solution? Can anyone confirm it?

JJHu1993, 2017-06-02:
I have modified the script to support new kernel(4.8.0-xx).
## link: https://github.com/JJHu1993/librealsense-patch.git

 ```
Patched kernel module created successfully

Unloading module uvcvideo
Applying patched module uvcvideo
Inserting uvcvideo succeeded

Script has completed successfully. Please consult the installation guide for further instruction.

jjhu@jjhu-computer:~/projects/librealsense$ uname -r
4.8.0-52-generic

```

@Ryanliyikun @hiptechboy @jamessergeant @mindThomas @tdnet12434 @Thanh-Binh

thesidjway, 2017-06-03:
Hello @JJHu1993 please commit the file patch-utils.sh . Thanks!

thesidjway, 2017-06-03:
Hello @JJHu1993 ,
Thanks a lot for working on the patch.
Could you please push the file patch-utils.sh as well? Thanks again.

fxprime, 2017-06-03:
@thesidjway You can replace that patch to old one in https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-xenial.sh
Thank you @JJHu1993 it worked for me :)

thesidjway, 2017-06-03:
@fxprime Thanks a lot!

Ender1618, 2017-06-04:
Hmm tried this, the script said it succeeded,

but the result of 
sudo dmesg | tail -n 50

results in:
[ 6861.673072] uvcvideo: Unknown symbol media_entity_create_link (err 0)
[ 6861.673128] uvcvideo: Unknown symbol media_entity_init (err 0)
[ 6861.673162] uvcvideo: Unknown symbol media_entity_cleanup (err 0)
[ 8490.871704] uvcvideo: Unknown symbol media_entity_create_link (err 0)
[ 8490.871773] uvcvideo: Unknown symbol media_entity_init (err 0)
[ 8490.871806] uvcvideo: Unknown symbol media_entity_cleanup (err 0)
[ 8919.318059] uvcvideo: Unknown symbol media_entity_create_link (err 0)
[ 8919.318201] uvcvideo: Unknown symbol media_entity_init (err 0)
[ 8919.318285] uvcvideo: Unknown symbol media_entity_cleanup (err 0)
 
for the uvcvideo module

I guess it didn't work, even though the script succeeded?

JJHu1993, 2017-06-04:
It worked for me and people around me whose kernel version is 4.8.xx

发自我的 iPhone

JJHu1993, 2017-06-04:
It worked for me and people around me whose kernel version is 4.8.xx
see pictures
## result
[image]
[image]

@Ender1618

thesidjway, 2017-06-05:
Hello, 
I have an Intel Aero setup with the vision accessory kit. I'm trying to run the camera on ROS (Ubuntu 16.04, Linux 4.4.0-78). For me, the 4.4 patch finishes successfully but neither am I able to see a /dev/video device connected to my system, nor do any of the example codes in the librealsense directory work. Has anyone of you had success using the realsense vision accessory kit with Ubuntu?

https://software.intel.com/en-us/aero/vision-kit

@JJHu1993 @fxprime @ev-mp @mindThomas

ev-mp, 2017-06-06:
Hi, is the R200 properly connected ? There are no UVC-related info in the screenshot. 
Additionally, please specify "lsusb" log

Atlas7, 2017-06-12:
@JJHu1993 - I just merged your revised patch script `scripts/patch-realsense-ubuntu-xenial.sh` to the librealsense repository locally on my machine. It worked! (just run an example cpp tutorial 2 stream script with the R200 plugged in.)

Log:

Kernal 4.8.x...

```
johnny@johnny-XPS-8700:~/repos/librealsense$ uname -r
4.8.0-54-generic
```

last few lines says new driver registered...

```
johnny@johnny-XPS-8700:~/repos/librealsense$ sudo dmesg | tail -n 50
.... (bla bla bla) ...
[  484.018926] media: Linux media interface: v0.10
[  484.038903] Linux video capture interface: v2.00
[  484.061164] usbcore: registered new interface driver uvcvideo
[  484.061165] USB Video Class driver (1.1.1)
```

Plugged the R200 camera in and run an example stream script - worked! :)

```
johnny@johnny-XPS-8700:/usr/local/bin$ ./cpp-tutorial-2-streams 
There are 1 connected RealSense devices.

Using device 0, an Intel RealSense R200
    Serial number: 2441000745
    Firmware version: 1.0.72.10
```

(Here is the revised librealsense branch with your revised script - https://github.com/Atlas7/librealsense/tree/patch-JJHu1993/scripts)

I guess you are good to go raising the pull request to merge to master?

Thanks a lot again!

Cheers
Johnny

JJHu1993, 2017-06-13:
Thanks for notice. I will write a script to support both kernel versions and pull it to master.

eric-schleicher, 2017-07-05:
So it's not clear from closing comment...   as of 7/5/2017 has this been pulled into master? or the best place to get this (working) is from @Atlas7's repo?

ev-mp, 2017-08-10:
As of Aug 10th, the librealsense patch is updated to support Ubuntu LTS 16.04.1 and 16.04.2 with kernels  4.4, 4.8* and 4.10.
*Starting from Aug 2nd, Ubuntu 16.04.2 users will receive kernel 4.10 version on their next system upgrade. After the upgrade it is important to [reapply the UVC driver patch](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation) as kernel 4.10 implements a more strict policy on handling UN-authorized kernel modules

closed 2017-08-10 by ev-mp