realsenseai/librealsense #445: Cmake Error: GLFW variables notfound

lutein, 2017-02-22:
I use Ubuntu 14.0.4 (Kernel: 4.4.0-57) and install according to the instruction. 
When I run `cmake ../ -DBUILD_EXAMPLES=true`, errors occur as following, while running `cmake ../` no error found.
~~~
-- Info: REALSENSE_VERSION_STRING=1.12.1
-- Building in a ROS environment
-- Using CATKIN_DEVEL_PREFIX: /home/minty/librealsense/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/minty/librealsense/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GLFW_INCLUDE_DIR
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
   used as include directory in directory /home/minty/librealsense/examples
GLFW_LIBRARIES
    linked by target "c-tutorial-1-depth" in directory /home/minty/librealsense/examples
    linked by target "c-tutorial-2-streams" in directory /home/minty/librealsense/examples
    linked by target "c-tutorial-3-pointcloud" in directory /home/minty/librealsense/examples
    linked by target "cpp-alignimages" in directory /home/minty/librealsense/examples
    linked by target "cpp-callback" in directory /home/minty/librealsense/examples
    linked by target "cpp-callback-2" in directory /home/minty/librealsense/examples
    linked by target "cpp-capture" in directory /home/minty/librealsense/examples
    linked by target "cpp-config-ui" in directory /home/minty/librealsense/examples
    linked by target "cpp-enumerate-devices" in directory /home/minty/librealsense/examples
    linked by target "cpp-headless" in directory /home/minty/librealsense/examples
    linked by target "cpp-motion-module" in directory /home/minty/librealsense/examples
    linked by target "cpp-multicam" in directory /home/minty/librealsense/examples
    linked by target "cpp-pointcloud" in directory /home/minty/librealsense/examples
    linked by target "cpp-restart" in directory /home/minty/librealsense/examples
    linked by target "cpp-stride" in directory /home/minty/librealsense/examples
    linked by target "cpp-tutorial-1-depth" in directory /home/minty/librealsense/examples
    linked by target "cpp-tutorial-2-streams" in directory /home/minty/librealsense/examples
    linked by target "cpp-tutorial-3-pointcloud" in directory /home/minty/librealsense/examples

-- Configuring incomplete, errors occurred!
See also "/home/minty/librealsense/build/CMakeFiles/CMakeOutput.log".
See also "/home/minty/librealsense/build/CMakeFiles/CMakeError.log".

Cee, 2017-03-16:
Same problem on Raspberry Pi 3.

ddiakopoulos, 2017-03-16:
@Cee just FYI - Raspberry Pi 3 does not have a USB 3.0 host controller so you cannot use any librealsense-supported cameras fundamentally

Cee, 2017-03-16:
@ddiakopoulos Thanks a lot!

thomasck, 2017-05-23:
same problem on the UP-board 
[ 61%] Building CXX object examples/CMakeFiles/cpp-config-ui.dir/cpp-config-ui.cpp.o
/home/tck/librealsense/examples/cpp-config-ui.cpp: In function ‘void show_message(GLFWwindow*, const string&, const string&)’:
/home/tck/librealsense/examples/cpp-config-ui.cpp:272:20: error: ‘GLFW_FLOATING’ was not declared in this scope
     glfwWindowHint(GLFW_FLOATING, GL_TRUE);
                    ^
examples/CMakeFiles/cpp-config-ui.dir/build.make:54: recipe for target 'examples/CMakeFiles/cpp-config-ui.dir/cpp-config-ui.cpp.o' failed
make[2]: *** [examples/CMakeFiles/cpp-config-ui.dir/cpp-config-ui.cpp.o] Error 1
CMakeFiles/Makefile2:420: recipe for target 'examples/CMakeFiles/cpp-config-ui.dir/all' failed
make[1]: *** [examples/CMakeFiles/cpp-config-ui.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

helpp

jlgarcia75, 2017-10-07:
For best support, post issues in the RealSense Community.

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

thomasck, 2017-10-07:
Hi Jesus,

I have resolved this issue. The solution was to run one of the install files from the scripts folder.
Thanks for the support!

Thomas

closed 2017-10-16 by jlgarcia75

codeAibee, 2018-03-16:
@thomasck  thanks! your method works! actually run script/install_glfw3.sh

vasiliyx, 2018-11-05:
> Hi Jesus,
> 
> I have resolved this issue. The solution was to run one of the install files from the scripts folder.
> Thanks for the support!
> 
> Thomas

Which file?
Same problem trying to get running Python3 on UpBoard Ubunty14.04 with realsense2