realsenseai/librealsense #447: SR300 is basically unusable with latest version

drhalftone, 2017-03-01:
| Required Info  |   |
|---|---|
| Camera Model | F200 | 
| Firmware Version | ???  | 
| Operating System & Version | Mac OS X 10.12.3  |
| Kernel Version (Linux Only)  |   |
| Build System |  Qt Creator and clang x64 | 

I used earlier versions of librealsense and connected to my Creative webcam without issue, but since updating to the latest version of librealsense, I can no longer get past the command:

            // GET SENSOR SIZE AND FRAME RATE FOR COLOR VIDEO
            rs_format format;
            rs_get_stream_mode(camera, colorStream, 0, (int *)&numColorCols, (int *)&numColorRows, &format, &frameRate, &err);

I am able to get the device count, and I am able to create the camera context.  I am able to read the RS_CAMERA_INFO_DEVICE_NAME and RS_CAMERA_INFO_DEVICE_SERIAL_NUMBER, but nothing else.

dorodnic, 2017-03-01:
Do you know what was the last version that worked for you? 
Also, what is the error you are getting?

drhalftone, 2017-03-03:
Looking at my github logs, the first time I got my software to work with the RS300 was Dec. 13, 2016.  I have been using the same version of librealsense since, until I recently ran into the choppy RGB video a week or two ago (I posted an issue).  I then had a few days trying to get the library compiled.  Now that is worked out, and I am able to talk to my R200, which I just purchased, but the SR300 and F200 are now not working at all past the line above.  

How do I read the error value?  The rs.h only defines rs_error as a structured, but it doesn't define its contents.

drhalftone, 2017-03-05:
Man, you nailed it right on the head. Looking at the err structure, I found the error class in rs.hpp and get the error string, and here it happened a few instructions above when I was reading the model name, which the RS300 doesn't support.  What surprised me is that I wasn't checking the err structure for those few lines, so it carried over to the later call.  Now my question is how does one clear the err structure so I can use it over and over again even after an error occurs like it did here?

closed 2017-03-05 by drhalftone