realsenseai/librealsense #450: Possibility of obtaining RAW image from camera sensor?

jamessergeant, 2017-03-03:
For any camera model but preferably the SR300, is it possible to obtain the RAW image from the camera sensor?

jamessergeant, 2017-03-05:
Closing issue as I found the RAW16 stream, on the SR300. Thanks.

jamessergeant, 2017-03-07:
Re-opening as closer inspection of the code shows that RAW formats seem to exist but not RAW streams.

ev-mp, 2017-03-07:
Format is a property of a stream, just as resolution or fps.
It is set, along with the rest of parameters, with _dev.enable_stream(...)_  ,or _rs_enable_stream(...)_  if you're in C.

It is not clear though, what do you mean for RAW: the RGB sensor is capable to provide RAW (Bayer) pixels; for Depth sensor, the meaning of RAW in most cases is Infrared images.

The list of supported formats per camera model is documented [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/supported_video_formats.pdf).

jamessergeant, 2017-03-07:
@ev-mp I'm after the RGB (bayer) raw. Unfortunately I haven't been able to achieve this yet. When I run cpp-enumerated-devices there are now raw formats under the color streams. This is for the SR300 (will update this message with firmware version and output of cpp-enumerated-devices tomorrow). Any advice on moving forward?

jamessergeant, 2017-03-07:
Output of the cpp-enumerate-devices example. Firmware version 3.17.0.0 and no raw streams under color.

```
Device 0 - Intel RealSense SR300:
 Serial number: 614205004058
 Firmware version: 3.17.0.0
 USB Port ID: 2-2-1
 Camera info: 
    DEVICE_NAME         : 	Intel RealSense SR300
    DEVICE_SERIAL_NUMBER: 	614205004058
    CAMERA_FIRMWARE_VERSION: 	3.17.0.0
 Supported options:                                    min        max       step  default  
    COLOR_BACKLIGHT_COMPENSATION                       : 0    ... 4           1     0         
    COLOR_BRIGHTNESS                                   : -64  ... 64          1     0         
    COLOR_CONTRAST                                     : 0    ... 100         1     50        
    COLOR_EXPOSURE                                     : 39   ... 10000       1     156       
    COLOR_GAIN                                         : 0    ... 128         1     64        
    COLOR_GAMMA                                        : 100  ... 500         1     300       
    COLOR_HUE                                          : -180 ... 180         1     0         
    COLOR_SATURATION                                   : 0    ... 100         1     64        
    COLOR_SHARPNESS                                    : 0    ... 100         1     50        
    COLOR_WHITE_BALANCE                                : 2800 ... 6500        10    4600      
    COLOR_ENABLE_AUTO_EXPOSURE                         : 0    ... 1           1     1         
    COLOR_ENABLE_AUTO_WHITE_BALANCE                    : 0    ... 1           1     1         
    F200_LASER_POWER                                   : 0    ... 16          1     16        
    F200_ACCURACY                                      : 1    ... 3           1     1         
    F200_MOTION_RANGE                                  : 0    ... 220         1     9         
    F200_FILTER_OPTION                                 : 0    ... 7           1     5         
    F200_CONFIDENCE_THRESHOLD                          : 0    ... 15          1     3         
    SR300_AUTO_RANGE_ENABLE_MOTION_VERSUS_RANGE        : 0    ... 2           1     -1        
    SR300_AUTO_RANGE_ENABLE_LASER                      : 0    ... 1           1     -1        
    SR300_AUTO_RANGE_MIN_MOTION_VERSUS_RANGE           : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_MAX_MOTION_VERSUS_RANGE           : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_START_MOTION_VERSUS_RANGE         : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_MIN_LASER                         : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_MAX_LASER                         : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_START_LASER                       : -32768... 32767       1     -1        
    SR300_AUTO_RANGE_UPPER_THRESHOLD                   : 0    ... 65535       1     -1        
    SR300_AUTO_RANGE_LOWER_THRESHOLD                   : 0    ... 65535       1     -1        
    FRAMES_QUEUE_SIZE                                  : 1    ... 20          1     20        
    HARDWARE_LOGGER_ENABLED                            : 0    ... 1           1     0         
 Stream COLOR - 130 modes:
  1920	x 1080	@ 30Hz	YUYV	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 30Hz	RGB8	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 30Hz	BGR8	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 30Hz	RGBA8	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 30Hz	BGRA8	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 10Hz	YUYV	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 10Hz	RGB8	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 10Hz	BGR8	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 10Hz	RGBA8	69.3 x 42.5 degrees, distortion = NONE
  1920	x 1080	@ 10Hz	BGRA8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 60Hz	YUYV	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 60Hz	RGB8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 60Hz	BGR8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 60Hz	RGBA8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 60Hz	BGRA8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 30Hz	YUYV	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 30Hz	RGB8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 30Hz	BGR8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 30Hz	RGBA8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 30Hz	BGRA8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 10Hz	YUYV	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 10Hz	RGB8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 10Hz	BGR8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 10Hz	RGBA8	69.3 x 42.5 degrees, distortion = NONE
  1280	x 720	@ 10Hz	BGRA8	69.3 x 42.5 degrees, distortion = NONE
  960	x 540	@ 60Hz	YUYV	69.3 x 42.5 degrees, distortion = NONE
  960	x 540	@ 60Hz	RGB8	69.3 x 42.5 degrees, distortion = NONE
  960	x 540	@ 60Hz	BGR8	69.3 x 42.5 degrees, distortion = NONE
[... 97 lines cut ...]
  320	x 180	@ 10Hz	YUYV	69.3 x 42.5 degrees, distortion = NONE
  320	x 180	@ 10Hz	RGB8	69.3 x 42.5 degrees, distortion = NONE
  320	x 180	@ 10Hz	BGR8	69.3 x 42.5 degrees, distortion = NONE
  320	x 180	@ 10Hz	RGBA8	69.3 x 42.5 degrees, distortion = NONE
  320	x 180	@ 10Hz	BGRA8	69.3 x 42.5 degrees, distortion = NONE
 Stream INFRARED - 18 modes:
  640	x 480	@ 200Hz	Y8	68 x 53.7 degrees, distortion = INVERSE_BROWN_CONRADY
[... cut ...]
  640	x 240	@ 110Hz	Y16	68 x 53.7 degrees, distortion = INVERSE_BROWN_CONRADY
  640	x 240	@ 60Hz	Y8	68 x 53.7 degrees, distortion = INVERSE_BROWN_CONRADY
  640	x 240	@ 60Hz	Y16	68 x 53.7 degrees, distortion = INVERSE_BROWN_CONRADY
  640	x 240	@ 30Hz	Y8	68 x 53.7 degrees, distortion = INVERSE_BROWN_CONRADY
  640	x 240	@ 30Hz	Y16	68 x 53.7 degrees, distortion = INVERSE_BROWN_CONRADY
  640	x 240	@ 10Hz	Y8	68 x 53.7 degrees, distortion = INVERSE_BROWN_CONRADY
  640	x 240	@ 10Hz	Y16	68 x 53.7 degrees, distortion = INVERSE_BROWN_CONRADY
```

dorodnic, 2017-03-09:
I believe SR300 exposes Bayer as V4L2_PIX_FMT_RW16, but I don't think librealsense supports it. 
You can check if this format is available by using v4l-utils (`sudo apt-get install v4l-utils`):
v4l2-ctl --list-formats-ext -d /dev/video0
v4l2-ctl --list-formats-ext -d /dev/video1
Assuming it is exposed by the camera (this is a calibration format, so it might depend on a specific build and manufacturer), we can help you enable it in librealsense

jamessergeant, 2017-03-10:
Hi @dorodnic below is the output of the v4l2-ctl command you specified. My interpretation is that the raw is not available?

```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 960x540
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 848x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x480
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 424x240
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 320x240
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 320x180
			Interval: Discrete 0.017s (60.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
```

jlgarcia75, 2017-10-09:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).
Hello @jamessergeant,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by jlgarcia75

jamessergeant, 2017-10-22:
Hi @jlgarcia75 sorry only just found this. I no longer require this functionality, thanks for following up.