realsenseai/librealsense #453: unable to use sr300 with ros

arpit15, 2017-03-04:
For installation I used [librealsense](http://wiki.ros.org/librealsense#Installation)
Then to install realsense_camera I downloaded [source](https://github.com/intel-ros/realsense) and then did `catkin build realsense_camera`
I am getting the following error when running
`roslaunch realsense_camera sr300_nodelet_default.launch`

[ERROR] [1488664965.026732946]: Failed to load nodelet [/camera/driver] of type [realsense_camera/SR300Nodelet] even after refreshing the cache: MultiLibraryClassLoader: Could not create object of class type realsense_camera::SR300Nodelet as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1488664965.026777004]: The error before refreshing the cache was: Failed to load library /home/arpit/ros_wss/catkin_ws/devel/lib//librealsense_camera_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/arpit/ros_wss/catkin_ws/devel/lib//librealsense_camera_nodelet.so: undefined symbol: _ZN2cv6String10deallocateEv)
[FATAL] [1488664965.026887421]: Failed to load nodelet '/camera/driver` of type `realsense_camera/SR300Nodelet` to manager `nodelet_manager'
[camera/driver-2] process has died [pid 19956, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load realsense_camera/SR300Nodelet nodelet_manager depth:=depth color:=color ir:=ir ir2:=ir2 fisheye:=fisheye imu:=imu __name:=driver __log:=/home/arpit/.ros/log/0e9c30e6-0109-11e7-9687-000af70eb219/camera-driver-2.log].
log file: /home/arpit/.ros/log/0e9c30e6-0109-11e7-9687-000af70eb219/camera-driver-2*.log
^C[camera/nodelet_manager-1] killing on exit
Inconsistency detected by ld.so: dl-close.c: 762: _dl_close: Assertion `map->l_init_called' failed!
shutting down processing monitor...

| Required Info  |   |
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Ubuntu 14.04  |
| Kernel Version (Linux Only)  |  4.4.0-64-generic  |
 
Please let me know what am missing to let the sensor run on my system.

mdhorn, 2017-03-05:
Why not use the prebuilt ROS package for realsense_camera?
http://wiki.ros.org/realsense_camera#Installation

arpit15, 2017-03-06:
@mdhorn I tried using the prebuilt package but it didn't work. I keep getting the same error.

mdhorn, 2017-03-06:
Can you post the error you get using both the prebuilt librealsense for ROS and the realsense_camera package for ROS?
Make sure you are only source the /opt/ros/indigo/setup.bash, and not any local work space setup files which might alter you PATH or LD_LIBRARY_PATH.

arpit15, 2017-03-09:
@mdhorn I am sry for replying late. I ran system updates. After which everything is working fine. Thank you for all the support.

closed 2017-03-21 by ddiakopoulos