realsenseai/librealsense #454: Where is the centre of the R200?

mgserafi, 2017-03-08:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | N/A  |
| Kernel Version (Linux Only)  |   |
| Build System |  N/A | 

On the projection documentation, it says:

> Each stream of images provided by librealsense is also associated with a separate 3D coordinate space, specified in meters, with the coordinate [0,0,0] referring to the center of the physical imager

Could you please elaborate on what the 'center of the physical imager' refers to? For the r200, is this the center of the visible lense? Or the center of the physical object? An image would be really helpful.

ddiakopoulos, 2017-03-16:
@mgserafi - Data streams coming out of librealsense (either real or synthetic) can be expressed in several different coordinate systems. One example would be the depth image itself. The origin of this data, expressed in meters, is the center of the left IR camera sensor.

mgserafi, 2017-03-16:
Thanks. Is there documentation that specifies the origin of data for each stream?

ddiakopoulos, 2017-03-21:
Hi @mgserafi - The origin in Z is only really relevant for the depth camera. We provide intrinsic calibration in the library (principal point, focal length) for left/right IR along with the RGB camera, and extrinsic calibration between all 3 sensors.

closed 2017-03-21 by ddiakopoulos