realsenseai/librealsense #455: Using R200 and librealsense without USB 3.0

thomasck, 2017-03-13:
R200
Intel Edison
Yocto Linux

Hello all,

This may be a naive question, but I was wondering if it's possible to extract the "points" data from my R200 data stream using an Intel Edison micro-controller. I know that the Realsense R200 typically requires a usb 3.0 interface, but it would be VERY nice to be able to get a small amount of data (points data) without needing an expensive or heavy micro controller such as the Joule or Up-board.

I was hoping perhaps someone has already interfaced with the R200 without usb 3.0, or if not, that it at least might be possible.

Thanks,

Thomas

teknotus, 2017-03-13:
It's more than just not fast enough. The camera interface won't even
communicate on USB 2.0.

thomasck, 2017-03-13:
Ok I guess what I am looking for is a way to hack the usb 3.0 protocol to make it compatible with other protocols. Probably not easy though.

Thanks anyway teknotus

ddiakopoulos, 2017-03-16:
Hi @thomasck The RealSense cameras are USB 3.0 only, and there's no USB 3.0 => 2.0 translator.

closed 2017-03-16 by ddiakopoulos