realsenseai/librealsense #458: Obtaining actual exposure

borax00, 2017-03-18:
| Required Info  |   |
|---|---|
| Camera Model | R200  | 
| Firmware Version | 1.0.72.06  | 
| Operating System & Version | Debian stretch |
| Kernel Version (Linux Only)  |  4.8.0 |
| Build System |  Makefile | 

I'd like to know the actual exposure-time that was used for each image.  The API suggests that this is possible but it's not working for me.

There's an "rs::device::get_frame_timestamp()" function but not an "rs:::frame::get_frame_actual_exposure()" function (seems like it would make sense but it's not there).

OK, so by setting up a callback with "rs::device::set_frame_callback()" I can get an rs::frame structure which has "rs::frame::get_frame_metadata()".
But the following code indicates that no meta-data is supported:

void frameCallback( const rs::frame &frame )
{
    for ( int metaI=0; metaI < RS_FRAME_METADATA_COUNT; metaI++ )
    {
        rs_frame_metadata meta = (rs_frame_metadata)metaI;
        bool supports = frame.supports_frame_metadata(meta);
        cout<<frame.get_stream_type()<<" supports " << rs_frame_metadata_to_string(meta) << ": " << supports << endl;
    }
}

In all cases (IR/depth/color streams) I get e.g.:
INFRARED supports ACTUAL_EXPOSURE: 0
INFRARED supports ACTUAL_FPS: 0

Is there a way to read the actual exposure?

ddiakopoulos, 2017-03-21:
The answer is generally yes. One nice feature is that the R200 packs this data into the first few bytes of each image. 

I see that the actual exposure field for that data you are trying to query is only supported for the fisheye camera of the ZR300 module (https://github.com/IntelRealSense/librealsense/blob/master/src/zr300.cpp#L479)

R200 supports async readback of exposure through a separate UVC control, but from your query above it seems you want the per data. You're in luck (https://github.com/IntelRealSense/librealsense/blob/master/src/ds-private.h#L429) but I don't think this particular data was ever exposed at the API level, so you might have some hacking to do!

borax00, 2017-03-23:
Thanks for the reply.

As it turns out the auto exposure isn't giving me much joy [goes unstable under some conditions] so I think I have to implement my own anyway (and if I do my own then I know the exposure without having to read it back).

closed 2017-04-03 by dorodnic