realsenseai/librealsense #459: IOError: [errno 13] permission denied

szx0112, 2017-03-20:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version | ubuntu 16.04  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013, XCode or Makefile | 

I am still learning ubuntu and ROS, and I get a lot of errors when catkin_make the librealsense package, shown below, appreciate for any comment. Thanks.

-- ~~  traversing 2 packages in topological order:
-- ~~  - librealsense
-- ~~  - realsense_camera
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'librealsense'
-- ==> add_subdirectory(librealsense)
-- Building in a ROS environment
**CMake Error: Could not open file for write in copy operation** /home/ubuntu/jetsonbot/build/librealsense/catkin_generated/pkg.develspace.context.pc.py.tmp
**CMake Error: : System Error: Permission denied**
CMake Error at /opt/ros/kinetic/share/catkin/cmake/em_expand.cmake:5 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:325 (em_expand)
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  librealsense/CMakeLists.txt:26 (catkin_package)

**CMake Error: Could not open file for write in copy operation** /home/ubuntu/jetsonbot/devel/share/librealsense/cmake/librealsenseConfig.cmake.tmp
**CMake Error: : System Error: Permission denied**
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:382 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  librealsense/CMakeLists.txt:26 (catkin_package)

**CMake Error: Could not open file for write in copy operation** /home/ubuntu/jetsonbot/devel/share/librealsense/cmake/librealsenseConfig-version.cmake.tmp
**CMake Error: : System Error: Permission denied**
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:387 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  librealsense/CMakeLists.txt:26 (catkin_package)

**CMake Error: Could not open file for write in copy operation** /home/ubuntu/jetsonbot/build/librealsense/catkin_generated/pkg.installspace.context.pc.py.tmp
**CMake Error: : System Error: Permission denied**I
CMake Error at /opt/ros/kinetic/share/catkin/cmake/em_expand.cmake:5 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:444 (em_expand)
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  librealsense/CMakeLists.txt:26 (catkin_package)

**CMake Error: Could not open file for write in copy operation** /home/ubuntu/jetsonbot/build/librealsense/catkin_generated/installspace/librealsenseConfig.cmake.tmp
**CMake Error: : System Error: Permission denied**
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:505 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  librealsense/CMakeLists.txt:26 (catkin_package)

**CMake Error: Could not open file for write in copy operation** /home/ubuntu/jetsonbot/build/librealsense/catkin_generated/installspace/librealsenseConfig-version.cmake.tmp
**CMake Error: : System Error: Permission denied**
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:514 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  librealsense/CMakeLists.txt:26 (catkin_package)

**CMake Error: Could not open file for write in copy operation** /home/ubuntu/jetsonbot/build/librealsense/cmake_uninstall.cmake.tmp
**CMake Error: : System Error: Permission denied**
CMake Error at librealsense/CMakeLists.txt:210 (configure_file):
  configure_file Problem configuring file

-- +++ processing catkin package: 'realsense_camera'
-- ==> add_subdirectory(realsense/realsense_camera)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/catkin/cmake/order_paths.py", line 37, in <module>
    main()
  File "/opt/ros/kinetic/share/catkin/cmake/order_paths.py", line 32, in main
    with open(args.outfile, 'w') as fh:
**IOError: [Errno 13] Permission denied:** '/home/ubuntu/jetsonbot/build/realsense/realsense_camera/catkin_generated/ordered_paths.cmake'
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  
  execute_process(/home/ubuntu/jetsonbot/build/catkin_generated/env_cached.sh
  "/usr/bin/python" "/opt/ros/kinetic/share/catkin/cmake/order_paths.py"
  "/home/ubuntu/jetsonbot/build/realsense/realsense_camera/catkin_generated/ordered_paths.cmake"
  "--paths-to-order" "/home/ubuntu/jetsonbot/src/librealsense/include"
  "/opt/ros/kinetic/include" "/usr/include"
  "/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp"
  "/opt/ros/kinetic/include/opencv-3.2.0"
  "/opt/ros/kinetic/include/opencv-3.2.0/opencv" "/usr/include/pcl-1.7"
  "/usr/include/eigen3" "/usr/include/ni" "/usr/include/vtk-6.2"
  "/usr/include/aarch64-linux-gnu" "/usr/include/freetype2"
  "/usr/include/aarch64-linux-gnu/freetype2" "/usr/include/tcl"
  "/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent"
  "/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include"
  "/usr/lib/openmpi/include" "/usr/lib/openmpi/include/openmpi"
  "/usr/include/python2.7" "/usr/include/hdf5/openmpi" "/usr/include/jsoncpp"
  "/usr/include/libxml2" "--prefixes" "/home/ubuntu/jetsonbot/devel"
  "/home/ubuntu/jetsonbot/src" "/opt/ros/kinetic") returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/list_insert_in_workspace_order.cmake:29 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:108 (list_insert_in_workspace_order)
  realsense/realsense_camera/CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/jetsonbot/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/jetsonbot/build/CMakeFiles/CMakeError.log".
Makefile:1760: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

szx0112, 2017-03-20:
The problem is solved by giving permission of librealsense and realsense to users. The downloaded package are originally owned by root

closed 2017-03-20 by szx0112

Vivoslespagnos, 2020-01-31:
Hi, I have the same problem, bellow my terminal copy : 

crabe@PRJ11:~/catkin_ws$ catkin_make --only-pkg-with-deps ultimate
Base path: /home/crabe/catkin_ws
Source space: /home/crabe/catkin_ws/src
Build space: /home/crabe/catkin_ws/build
Devel space: /home/crabe/catkin_ws/devel
Install space: /home/crabe/catkin_ws/install
Whitelisted packages: ultimate
####
#### Running command: "make cmake_check_build_system" in "/home/crabe/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/crabe/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/crabe/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/crabe/catkin_ws/devel;/opt/ros/melodic
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- Using CATKIN_WHITELIST_PACKAGES: ultimate
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - ultimate
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'ultimate'
-- ==> add_subdirectory(ultimate)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
Traceback (most recent call last):
  File "/opt/ros/melodic/share/catkin/cmake/order_paths.py", line 35, in <module>
    main()
  File "/opt/ros/melodic/share/catkin/cmake/order_paths.py", line 28, in main
    os.makedirs(outdir)
  File "/usr/lib/python2.7/os.py", line 157, in makedirs
    mkdir(name, mode)
OSError: [Errno 13] Permission denied: '/catkin_generated'
CMake Error at /opt/ros/melodic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/crabe/catkin_ws/build/catkin_generated/env_cached.sh
  "/usr/bin/python2" "/opt/ros/melodic/share/catkin/cmake/order_paths.py"
  "/catkin_generated/ordered_paths.cmake" "--paths-to-order"
  "/opt/ros/melodic/include" "/usr/include" "/usr/include/opencv"
  "/opt/ros/melodic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp"
  "--prefixes" "/home/crabe/catkin_ws/devel" "/home/crabe/catkin_ws/src"
  "/home/crabe/catkin_ws/devel" "/home/crabe/catkin_ws/src"
  "/opt/ros/melodic") returned error code 1
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/list_insert_in_workspace_order.cmake:29 (safe_execute_process)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:108 (list_insert_in_workspace_order)
  ultimate/CMakeLists.txt:67 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/crabe/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/crabe/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:588: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

Did you find something, I dont know how to give permission to users !