realsenseai/librealsense #463: [Question] Use Manual Calibration Values for Registration

Maghoumi, 2017-03-25:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.10 | 
| Operating System & Version |  Windows 10 |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2015 | 

I need to know if it is possible to use self-calibrated values with librealsense (or modify it in a way that would use the supplied calibration matrices, not the ones stored on the device itself).
For my application, a very precise calibration is needed because it is important to get very good registration between the depth and the RGB data. The current data I'm getting does not look very good around the edges of the objects.

As I understand it, the depth is a stream of its own, computed with the ASIC on the device itself (much like Kinect's depth stream). However, with the Kinect, finding the extrinsics between the RGB and the virtual depth camera is straightforward, as you can do a standard stereo calibration between the IR camera and the RGB camera. The extrinsics found between the IR and the RGB are the same as the extrinsics between the RGB and the virtual depth camera.

On my R200, I can see that the extrinsics between the virtual depth camera and "Infrared1" is identity (that is, it is assumed that the virtual depth camera is located at the position of "infrared1"). However, there is a non-identity extrinsics matrix between "infrared1" and "infrared2". My first question is, can I assume this is by design and the virtual depth camera is always taken to be at the position of "infrared1" physical camera?

Now my second question: is there a way to tell the camera to use my manual calibration values for creating the depth stream? (As in saying "Hey R200, could you not use what you have stored on your device memory and use what I give you?") 😄 

I tried the calibration tool on Intel's website. It tells me my calibration is fine, but I based on what I saw, I think the tool is a bit sketchy and is doing a similar thing to the plane fitting approach of Daniel Herrera (from the Kinect calibration [toolbox](https://www.mathworks.com/matlabcentral/linkexchange/links/2882-kinect-calibration-toolbox)), and based on my experience, that method sucks badly for my particular application and does not produce a good enough result.

Any help is appreciated!

Caopi, 2017-03-27:
I am currently using the SR300 and I am also experiencing problems with the alignment between rgb and depth around the edges of objects, which I suspect to be calibration problems. 

I would like to recalibrate the camera to see if this makes any difference, but even the official tool does not support the SR300. This tool was the suggested solution for a similar issue with the R200 - https://github.com/IntelRealSense/librealsense/issues/323. Also, as Maghoumi said, the calibration method of the tool might not even be a good idea for certain applications.

Please correct me if I am missing something, but with librealsense also not offering any possibility to do this, there seems to be no option to change the SR300 camera's calibration?

If that is the case, I would second a feature request for API support to change the calibration of the Realsense, since I think this is really an essential feature that this library should have.

A tool to actually do the calibration isn't even necessary in a first implementation. I have done calibration for the IR1 and RGB image using OpenCV already, so values would be there, just no possibility to actually use them.

Maghoumi, 2017-03-27:
Yeah... Interestingly enough there was a mention of "dynamic calibration" in #323, which probably means that rewriting factory data is possible (and judging by the fact that the calibration app can modify the values stored in the device, I'd think the write functionality can be accessed via software).

If the ASIC is doing all the computations using the device stored calibration data, then there's probably no other way than modifying that data to use manual calibration values...

It would be great if one of the developers could comment on this.

ddiakopoulos, 2017-03-28:
Hi @Maghoumi - many of your assumptions are correct. The depth map is generated in the coordinate system of infrared1 (what we call the 'left' imager); therefore the extrinsic parameters of infrared1 and depth are the same. From this perspective, the process of extrinsic calibration between depth and RGB is no different from the Kinect.

The R200 is calibrated in factory and uses these values for depth map computation on the ASIC. The RGB sensor was placed on the board as close as possible to infrared1 to minimize depth/rgb rectification errors. However, if you compare the lens between R200 and Kinect, you'd see that R200 was size-optimized. In rather unscientific terms, the Kinect lens is bigger and probably better and might have more uniform properties that are amenable to the errors you're seeing at the edges.

The SR300 -- last time I checked -- does not have per-unit in-factory calibration and relies on an averaged model of the lens. 

> If that is the case, I would second a feature request for API support to change the calibration of the Realsense, since I think this is really an essential feature that this library should have.

We've debated it numerous times internally. The factory calibration toolset is huge -- easily 100k lines of C++ just for R200 alone. It's not maintainable as an open-source project, so it will never be included as part of librealsense.

> A tool to actually do the calibration isn't even necessary in a first implementation. I have done calibration for the IR1 and RGB image using OpenCV already, so values would be there, just no possibility to actually use them.

If you require better intrinsic/extrinsic calibration between left imager and depth, manual calibration and rectification is entirely possible and recommended. IR1 and RGB rectification is not performed on chip, so there's no barriers to using this depth/RGB mapping.

Maghoumi, 2017-03-28:
Thanks for the clarifications @ddiakopoulos. 

> We've debated it numerous times internally. The factory calibration toolset is huge -- easily 100k lines of C++ just for R200 alone. It's not maintainable as an open-source project, so it will never be included as part of librealsense.

Would there be at least the possibility of adding a capability to load and store manual calibration on the device in the RealSense calibration app? As I said, I don't trust what I see with the calibration app as it currently is... I'm not even sure if the depth distortion model of the RealSense devices is similar to that of Kinect, so for many applications, there's just not enough information about the underworkings of the sensor so that we can proceed with building our applications using sound assumptions. (I would not have known that some RealSense cameras had a barometer if I hadn't seen #4!!)
Also, to me, a quick calibration of the sensor using a planar surface does not sound like an accurate calibration suitable for precise applications at all : ) It's more like a DIY@Home calibration geared towards casual/home users.
I would argue that since RealSense devices are targeted towards developers, the developers would appreciate every bit of freedom you can grant them.

Thanks again for your response.

Caopi, 2017-03-28:
> We've debated it numerous times internally. The factory calibration toolset is huge -- easily 100k lines of C++ just for R200 alone. It's not maintainable as an open-source project, so it will never be included as part of librealsense.

Really sad to hear this, but thanks for the clarification.

> If you require better intrinsic/extrinsic calibration between left imager and depth, manual calibration and rectification is entirely possible and recommended. IR1 and RGB rectification is not performed on chip, so there's no barriers to using this depth/RGB mapping.

That's what I was planning to do earlier. Then I discovered librealsense's capabilities to do rectification, registration and projections, so I was hoping it could save me some work. 

But, with the calibration being off, I guess all these features just won't be usable with the required accuracy, that's a shame.

ddiakopoulos, 2017-03-28:
@Caopi this is an inaccurate assessment. Most rectification, registration, and projection is implemented on the CPU side so you can always see what's going on and re-implement the same functions "above" the library using your own calibration parameters. Only the rectification of IR1/IR2 and block matching is done on the ASIC. I don't believe there's any barriers to improving the accuracy of RGB/depth registration using your own calibration.

Caopi, 2017-03-28:
@ddiakopoulos Ok that's true and we need to differentiate between the different calibrations on ASIC and software/library. 

But this raises the question: if it's as simple as using the exact same code and only replacing the calibration parameters that the library uses, why doesn't the library offer this as a feature?

Copying code and rebuilding the library just to change hardcoded parameters isn't really what I would expect from a library. Specifically for RGB/depth registration, this would greatly increase the usability of all the already existing librealsense features.

iebeid, 2017-04-17:
Hi guys,
I've been trying to get an undistorted color image from the R200 camera and I did after i calibrated and found the camera matrix and the distortion parameters, however my question is,
is the depth frame distorted as well, or the R200 provides an undistorted depth frame, I read your comments, and also tried to post on intel community but I did not get an answer.

zhangcuthand, 2017-04-19:
I am a realsense R200 user, and reserching for VSLAM based on RGB-D camera SLAM, but I the calibration which Intel applied in Windows so poor for developing algorithm Precisely on based my PC

kazoo-kmt, 2017-08-08:
@ddiakopoulos 
Please let me confirm the current status about SR300. 
- Intrinsic parameters for both color camera and IR camera (=3D camera) are already calibrated by Intel when the SR300 is manufactured.
- It's not possible to change/modify these intrinsic parameters. (Or, is it possible to change by using SDK?)
- To calibrate the extrinsic parameters (R, T) for the color camera, we can use OpnCV's calibration method + checker board. While this extrinsic parameters' calibration, need to give/use the intrinsic parameters given by Intel.
- Once after getting the extrinsic parameters of the color camera, we can calculate the extrinsic parameters for IR camera by using the transformation matrix between the color and the IR camera, which we can get by calling `get_extrinsics`.

jlgarcia75, 2017-11-08:
@kazoo-kmt,

Your assumptions are correct.

Regards,
Jesus G.
Intel Customer Support