realsenseai/librealsense #465: add support for RS_DISTORTION_FTHETA into rs_project_point_to_pixel()

melax, 2017-03-27:
Using the Intel RealSense ZR300  that has the "motion module" with fish_eye camera.  Accessing the fish_eye stream and getting the intrinsic and distortion parameters is supported by the current librealsense.   Additionally there is a sample displaying these newly available streams.  

However, there is no documentations on specifically what the fisheye intrinsic parameters mean.   At minimum, if the equations were simply added to the function rs_project_point_to_pixel().  Currently the function asserts when provided intrinsics returned from the api.

```
  auto intrinsics = dcam.dev->get_stream_intrinsics(rs::stream::fisheye);
  auto p = intrinsics.project(rs::float3{ my_x, my_y, my_z });   // this will assert!!!!!
```

We tried an radial projection with distortion implementation and a few variants based on common conventions that people tend to use (there are a number of them), but in our tests things didn't quite line up.  Consequently, not sure if equation is correct, or if the calibration parameters are off.  code tested is:
```
static void rs_project_point_to_pixel(float pixel[2], const struct rs_intrinsics * intrin, const float point[3])
{
    assert(intrin->model != RS_DISTORTION_INVERSE_BROWN_CONRADY); // Cannot project to an inverse-distorted image
    // comment out the assert for FTHETA:
    //assert(intrin->model != RS_DISTORTION_FTHETA); // Cannot project to an ftheta image

    float x = point[0] / point[2], y = point[1] / point[2];

    if(intrin->model == RS_DISTORTION_MODIFIED_BROWN_CONRADY) 
    {
        .... same as before ...  
    }
    if (intrin->model == RS_DISTORTION_FTHETA)
    {
		float r2 = x*x + y*y;
		float r = sqrt(x*x + y*y);
		float t = atan(r);
		float f = 1 + intrin->coeffs[0] * t*t;    // we noticed other coeffs, besides coeffs[0], were all 0.0 anyways
		x *= f;
		y *= f;
		if (r>1e-8)
		{
			x *= t / r;
			y *= t / r;
		}
    }
    pixel[0] = x * intrin->fx + intrin->ppx;
    pixel[1] = y * intrin->fy + intrin->ppy;
}
```
Note, the above code may be incorrect.  Anybody using the fish_eye data will need to know the correct equation.  Library on github should be updated with the necessary code added to this function in rsutil.h.   

I'd also suggest adding such correspondence to the multi-stream viewer sample.   i.e. when the mouse cursor is over a point in one of the stream images, the program also displays a small highlight over all the other streams that corresponds to the same point if one exists and can be computed.  This would make it very clear how to apply the intrinsics and extrinsics for each stream.   Such short code samples are a way to overcome any ambiguity or missing information in the documentation.  

| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |   | 
| Operating System & Version |  Windows   |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2015  | 

i'm using windows and vs2015, but this would also apply to other operating systems and compilers.

melax, 2017-03-27:
Someone suggested I try the distortion model from equation 13 from "Straight lines have to be be straight: automatic calibration and removal of distortion from scenes of structured environments" by Devernay and Faugeras.    https://hal.inria.fr/inria-00267247/document 

So i just tried the following implementation:
```
    float r = sqrt(x*x + y*y);
    auto rd = (1.0f / intrin->coeffs[0] * atan(2 * r* tan(intrin->coeffs[0] / 2.0f)));
    x *= rd / r;
    y *= rd / r;
```

seems closer and more consistent throughout the view volume of the depth camera, but it doesn't quite line up pixel perfect in my test program.   If someone could confirm whether this is how the zr300 fisheye intrinsics are supposed to be applied that would be nice.   even better, someone in the know please just add the correct equation to the code (whether the above snippet or something else).  

thanks in advance.

dorodnic, 2017-04-03:
Hi @melax,
To be honest, texture mapping between depth and fish-eye was never explicitly requested for ZR300 product so we never got to implementing it. There is no **official** documentation from Algo, but we do have the code used to undistort fish-eye frames:
[image]
It seems very close to the second method you suggested (it should be more or less the inverse).
And indeed using the second formula, fish-eye frames seem to be undistored and reasonably mapped to depth:
[image]
I'll try to get the official confirmation on this and would be happy to add this functionality to librealsense.

jlgarcia75, 2017-10-07:
For best support, post issues in the RealSense Community.

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

melax, 2017-10-07:
got it implemented.   we're all good.

closed 2017-10-07 by melax