realsenseai/librealsense #466: sr300 color stream does not work

volley84, 2017-03-28:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | Windows 10 pro |
| Kernel Version (Linux Only)  |   |
| Build System |  VS Enterprise 2015 update 3 (14.0.25431) |

ev-mp, 2017-03-28:
Hi, please check and provide the following:

HW:
- Did you verify that the RGB lens protection cap is removed?
- Is it a new camera that never been tested, or is it a new issue with an existing camera ?
- What is the FW version of SR300 ? (cpp-enumerate-devices)

SW:
- What Librealsense version you use?
- Are there any error/warning messages in the console when you run 'cpp-tutorial-2-streams' ?
- Did you try to isolate the issue and active streams one-by-one ? ('cpp-config-ui').
- Does 'cpp-headless' produces a valid (non-zero) snapshot for RGB stream ? If so - does it include black pixels only? 

There are no known issues that would render RGB sensor totally unusable, so we need data to establish the root cause of it.

volley84, 2017-03-29:
Thank for the reply. I am answer the following:

HW:

- Sorry, I am not know, but I am show image is: 
[image]
 The cap is removed? 
- New camera never been tested
- Fireware version: 3.21.0.0 

SW:

- Librealsense 1.12.1
- Probably,there are no errors in the console when you run 'cpp-tutorial-2-streams', but below screen:
[image]
- I did try 'cpp-config-ui' : 
https://youtu.be/DZG8Jlguj4A
- I did  'cpp-headless' produces a valid (non-zero) snapshot for RGB streams. Yes, does it include black pixels only
cpp-headless-output-COLOR.png
[image]
cpp-headless-output-DEPTH.png
[image]
cpp-headless-output-INFRARED.png
[image]

ev-mp, 2017-03-29:
Thanks for the info, indeed, there are no observable on protocol/application layer.
I'd ask you to perform additional test with 3rd party tool - run Windows10 builtin 'Camera' application to capture RGB channel of SR300 .
When executing, you can toggle between the available UVC streaming devices via the button at the upper left corner (see below)
[image]
This is the snapshot I just did with SR300 and 'Camera' application

volley84, 2017-03-30:
@ev-mp Thank for the reply. I toggled between the available UVC streaming devices Senz3D, F200 and SR300: https://youtu.be/bU25nPhQuN8

ev-mp, 2017-03-30:
Unfortunately now we've established that the issue is HW and not in Software Stack.
So you have to return the faulty SR300 to retailer for replacement.

It is still possible, though, that the issue could be resolved by upgrading/downgrading device Firmware with Intel official [RealSense SDK](https://software.intel.com/en-us/videos/downloading-and-installing-the-intel-realsense-sdk).
But based on all the tests and cross-verification performed, I'd describe this as very unlikely.

warmlayu, 2017-09-19:
Hi, guys,
Have you fixed the issue finally? I met the same issue with my SR300...
@volley84 Have you returned the camera for replacement? if yes, does the new camera work?
Thank you very much!

volley84, 2017-09-26:
Hi, @warmlayu !
My sr300 camera is already working, but not the quality compared to the F200 camera. Previously, the sr300 camera did not work, because windows 10 automatically downloaded the inappropriate driver for the camera. So I turned off the internet. I manually downloaded the driver from the official site, I installed it with my hands. And the camera works.

closed 2017-10-26 by jlgarcia75

TheElec, 2017-11-24:
I ran into the same troubles as @volley84, sadly his fix didn't work for me. And I'll have problems sending it back to the retailer, considering I had the camera shipped overseas.