realsenseai/librealsense #467: how to use distance from depth to avoid detection after a certain range

zubair123khan, 2017-03-28:
| Required Info  |   |
|---|---|
| Camera Model | R200|
| Firmware Version |  1.0.72.10 | 
| Operating System & Version | ubuntu 16.04   |
| Kernel Version (Linux Only)  | 4.4.0-66-generic |
| Build System |  VS2013, XCode or Makefile | 

hi guys i am using librealsense library in opencv , i have depth and rgb streams from camera, also i am able t detect red color blob ,, now i want to include distance factor from depth...
i want to avoid if there is one more red color after a certain distance .... 
how can i use depth to parameterize distance to track my blob ????

thanks all

zubair123khan, 2017-03-28:
`
#include <iostream>
#include "highgui.hpp"
#include "imgproc.hpp"
#include <librealsense/rs.hpp>

using namespace cv;
using namespace std;

 int main( int argc, char** argv )
 {
rs::context ctx;

// Access the first available RealSense device
rs::device * dev = ctx.get_device(0);

// Configure color stream to run at VGA resolution at 30 frames per second
dev->enable_stream(rs::stream::color, 640, 480, rs::format::bgr8, 30);

// We must also configure depth stream
dev->enable_stream(rs::stream::depth, 640, 480, rs::format::z16, 30);

// Start streaming
dev->start();

// Determine depth value corresponding to one meter
const uint16_t one_meter = static_cast<uint16_t>(1.0f / dev->get_depth_scale());

// Creating OpenCV Matrix from a color image
Mat color(Size(640, 480), CV_8UC3, (void*)dev->get_frame_data(rs::stream::color), Mat::AUTO_STEP);

// Creating OpenCV Matrix from a depth image
Mat depth(Size(640, 480), CV_16U, (void*)dev->get_frame_data(rs::stream::depth), Mat::AUTO_STEP);

float scale = dev->get_depth_scale();

// Display in a GUI
namedWindow("Display Image depth", WINDOW_AUTOSIZE );
namedWindow("Display Image", WINDOW_AUTOSIZE );
namedWindow("Control", CV_WINDOW_AUTOSIZE); //create a window called "Control"

 int iLowH = 170;
 int iHighH = 179;

 int iLowS = 150;
 int iHighS = 255;

 int iLowV = 60;
 int iHighV = 255;

 //Create trackbars in "Control" window
 createTrackbar("LowH", "Control", &iLowH, 179); //Hue (0 - 179)
 createTrackbar("HighH", "Control", &iHighH, 179);

 createTrackbar("LowS", "Control", &iLowS, 255); //Saturation (0 - 255)
 createTrackbar("HighS", "Control", &iHighS, 255);

 createTrackbar("LowV", "Control", &iLowV, 255);//Value (0 - 255)
 createTrackbar("HighV", "Control", &iHighV, 255);

 int iLastX = -1;
 int iLastY = -1;

while(cv::waitKey(30) <= 0)
{
	dev->wait_for_frames();
	if (!dev->is_streaming())
	{
		cout << "[ERROR] Disable streaming" << endl;
	}
	if (!dev->is_stream_enabled(rs::stream::color))
	{
		cout << "[ERROR] Disable color streaming" << endl;
	}

	color.data = (unsigned char*)dev->get_frame_data(rs::stream::color);
	depth.data = (unsigned char*)dev->get_frame_data(rs::stream::depth);
    imshow("Display Image", color);
    imshow("Display depth",depth);

    // Retrieve depth data, which was previously configured as a 640 x 480 image of 16-bit depth values
           const uint16_t * depth_frame = reinterpret_cast<const uint16_t *>(dev->get_frame_data(rs::stream::depth));
           int depth = *depth_frame++;

if (scale <10 && depth<one_meter)
{

 //Create a black image with the size as the camera output
 Mat imgLines = Mat::zeros(color.size(), CV_8UC3 );;
 cout<<"2"<<endl;

  Mat imgHSV;

  cvtColor(color, imgHSV, COLOR_BGR2HSV); //Convert the captured frame from BGR to HSV

  Mat imgThresholded;

  inRange(imgHSV, Scalar(iLowH, iLowS, iLowV), Scalar(iHighH, iHighS, iHighV), imgThresholded); //Threshold the image

  //morphological opening (removes small objects from the foreground)
  erode(imgThresholded, imgThresholded, getStructuringElement(MORPH_ELLIPSE, Size(5,5)) );
  dilate( imgThresholded, imgThresholded, getStructuringElement(MORPH_ELLIPSE, Size(5,5)) );

  //morphological closing (removes small holes from the foreground)
  dilate( imgThresholded, imgThresholded, getStructuringElement(MORPH_ELLIPSE, Size(5,5)) );
  erode(imgThresholded, imgThresholded, getStructuringElement(MORPH_ELLIPSE, Size(5,5)) );

  //Calculate the moments of the thresholded image
  Moments oMoments = moments(imgThresholded);

  double dM01 = oMoments.m01;
  double dM10 = oMoments.m10;
  double dArea = oMoments.m00;

  // if the area <= 10000, I consider that the there are no object in the image and it's because of the noise, the area is not zero
  if (dArea > 10000)
  {
   //calculate the position of the image
   int posX = dM10 / dArea;
   int posY = dM01 / dArea;
   if (iLastX >= 0 && iLastY >= 0 && posX >= 0 && posY >= 0)
     {
      //Draw a red line from the previous point to the current point
      line(imgLines, Point(posX, posY), Point(iLastX, iLastY), Scalar(0,0,255), 2);
     }

     iLastX = posX;
     iLastY = posY;

  //imshow("Thresholded Image", imgThresholded); //show the thresholded image

  color = color + imgLines;
//show the original image

  /*if (waitKey(30) == 27) //wait for 'esc' key press for 30ms. If 'esc' key is pressed, break loop
       {
            cout << "esc key is pressed by user" << endl;
            exit(1);
       }*/

      vector<Point> points;
        Mat_<uchar>::iterator it = imgThresholded.begin<uchar>();
        Mat_<uchar>::iterator end = imgThresholded.end<uchar>();
        for (; it != end; ++it)
          if (*it)
        points.push_back(it.pos());
        RotatedRect box = cv::minAreaRect(cv::Mat(points));

        double angle = box.angle;

        if (angle < -45.)
          angle += 90.;
        Point2f vertices[4];
          box.points(vertices);

          for(int i = 0; i < 4; ++i)
          {  line(imgThresholded, vertices[i], vertices[(i + 1) % 4], cv::Scalar(255, 0, 0), 1, CV_AA);
          }

          imshow("Result", imgThresholded);
          cout<<posX<<posY<<std::endl;

 }
}
}

   return 0;
}

`
this is the full code

zubair123khan, 2017-03-28:
@mdhorn

jlgarcia75, 2017-11-01:
Hello @zubair123khan,

Do you still need help with this issue?

Regards,
Jesus G.
Intel Customer Support

closed 2017-11-09 by jlgarcia75