realsenseai/librealsense #468: Issues while using two realsense

xiechangwu, 2017-04-01:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | R200={1.0.72.06}  SR300={3.17.0.0}  | 
| Operating System & Version |  Ubuntu 16.04  |
| Kernel Version (Linux Only)  | 4.4  for PC/3.10.96 for TX1 with uvc patch |
| Build System | Makefile | 

I'm using two realsense on my board(nvidia TX1), an R200 and an SR300. Open the librealsense exmaple cpp-multicam, the default resolution is 640X480, averything goes will.
In my application, we want use the 1080P resolution, so I change it and rebuild cpp-multicam.cpp(RGB:1080P, Depth:640x480)
At this time, the R200 RGB data occures incorrect,  see below:

Summary:
I tried using above two camera with my PC(thinkPad with two standalone USB3.0 port), this problem also not disappeared, but the R200 image is better than on TX1 board.
While I using two R200 with 1080P, this problem is the same.
While I using only one R200 with 1080P plus 640X480 depth on TX1 board, the R200 RGB data sometimes not correct, but much better than two realsense.
I've confirmed that the source image from librealsense is incorrect by saving the date after read date .
TX1 board kernel version 3.10.96 with UVC module pached
PC kernel version is 4.4 and installed with official guides
below is a correct picture for R200:

dorodnic, 2017-04-03:
This sounds like side-effects of USB buffer overflows. It happens whenever the host PC is not pulling USB buffers from the device fast enough. At the moment, librealsense just passes such partial frames to the application. 
This can be verified by printing out the value of `buf.bytesused` at [uvc-v4l2.cpp:344](https://github.com/IntelRealSense/librealsense/blob/master/src/uvc-v4l2.cpp#L344). 
During normal streaming this value should equal to `_buffers[buf.index].length`, but during overflows it would be lower. 
The overflows themselves are either caused by high CPU usage in parallel to streaming, or just a weak CPU (perhaps CPU is being thermal-throttled? In this case cooling can have big effect)
As a possible work-around I have previously recommended to increase the number of URBs in uvcvideo from 5 to about 8 (change `#define UVC_URBS 5` in `/drivers/media/usb/uvc/uvcvideo.h`, rebuild and replace the Kernel module). In theory this should increase the queue size between USB layer and UVC driver, reducing partial frames at expense of some extra latency. However, this might have other unexpected side-effects and would not be recommended in production environment. 

P.S. Just to make sure, you are using V4L back-end and not LIBUVC, correct?

xiechangwu, 2017-04-05:
Hi, dorodnic.
  Thanks very much for you help.

- I tried with your solution by changed UVC_URBS to 8/16 , UVC_MAX_VIDEO_BUFFERS to 64 but neither of them hss any effects for this probelm.

- Good news is, buf.bytesused and _buffers[buf.index].length were acts as you expected.
I add "LOG_WARNING("buf.bytesused=" << buf.bytesused << "  sub->buffers[buf.index].length:" << sub->buffers[buf.index].length);"  within uvc-v4l2.cpp:344. And the logs are as follows:

rs.warn: Subdevice 1 frame status 0x1
rs.warn: buf.bytesused=3932160  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=4147200  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=589064  sub->buffers[buf.index].length:589064
rs.warn: Subdevice 1 frame status 0x1
rs.warn: buf.bytesused=3953664  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=589064  sub->buffers[buf.index].length:589064
rs.warn: Subdevice 1 frame status 0x1
rs.warn: buf.bytesused=3910656  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=3459072  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=589064  sub->buffers[buf.index].length:589064
rs.warn: Subdevice 1 frame status 0x1
rs.warn: buf.bytesused=3867648  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=3437568  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=4147200  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=614400  sub->buffers[buf.index].length:614400
rs.warn: buf.bytesused=4147200  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=3932160  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=614400  sub->buffers[buf.index].length:614400
rs.warn: buf.bytesused=4147200  sub->buffers[buf.index].length:4147200
rs.warn: buf.bytesused=589064  sub->buffers[buf.index].length:589064

@dorodnic

xiechangwu, 2017-04-06:
@dorodnic 
Please note that this issue only appears for R200， the SR300 doesn't has this problem.
Is it caused by an additional picture called "infrared" when using R200?

xiechangwu, 2017-04-06:
[image]

dorodnic, 2017-04-06:
If you are enabling Infrared and Infrared2 on the R200 in your application, this would be the first thing to disable. However, it might be that your target platform simply struggles to handle two 1080P streams + depth data. What is the CPU utilization when the problem is happening? When you are requesting RGB data from librealsense, we are internally doing YUY to RGB conversion, and it is known to be very CPU intensive. Even if SSE instructions are available and enabled, this might be too much for some hosts to handle and keep up with the USB transfer. 
Could you try your test, but ask for YUY format instead of RGB in the sample App? (You could consider offloading YUY to RGB conversion to the GPU. We are not doing any of that inside librealsense to avoid adding OpenCL as a dependency, but I know it has been done before)

xiechangwu, 2017-04-08:
@dorodnic 

- I didn't enable  Infrared and Infrared2 on the R200.
- I tried with yuyv format and then convert it to rgb data by the use of opencv, but it has no effects(my opencv has been optimized with GPU)
- I'm not so agree with "CPU load intensive". While I only enable and show RGB data, the pictures are not so bad. But if I enabled depth stream(Note that I didn't show the depth data on screen, and didn't read any depth data), the pictures becomes very terrible.
- By the way, change camera fps to 10/15hz has a better results.

Any suggestions?
Thanks!

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by jlgarcia75