realsenseai/librealsense #471: auto exposure for zr300 fisheye stream not working

melax, 2017-04-03:
Auto exposure for the fisheye camera doesn't work if you are getting fisheye stream data the regular way.   

```
    // during setup, add fish eye streams to application:
    dev->set_option(rs::option::fisheye_color_auto_exposure, 1);
    dev->enable_stream(rs::stream::fisheye, 640, 480, rs::format::raw8, 60);
 ...
    // in main loop:
    auto data = (unsigned char*)dev->get_frame_data(rs::stream::fisheye);
```

Digging into the code, it looks like some sort of new callback api has been added, and the fish-eye auto exposure is only provided for applications that switch over from the regular librealsense api to use the new callback mechanisms.

```
    // in librealsense/src/device.cpp   around line #469:
                if (config.callbacks[streams[i]])
                {
                    ... 
                        on_before_callback(streams[i], frame_ref, archive);

    // in zr300.cpp:
    void zr300_camera::on_before_callback( ... )
    {
        if (!to_add_frames || stream != RS_STREAM_FISHEYE)
            return;

        auto_exposure->add_frame(clone_frame(frame), archive);
    }
```

| Required Info  |   |
|---|---|
| Camera Model | zr300 | 
| Firmware Version |   | 
| Operating System & Version |  Windows (this bug would apply to other OSs too) |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2015  (bug in the code and would apply to other compilers as well) |

souch55, 2017-06-14:
Hi melax addressing to yours issue i have an suggestion to handle auto_exposure with out call_backs my suggestion are changes in the source code,Plz do let me know if this changes works for you related to fish-eye auto exposure ,issue.

//Following addition in device.cpp
if (config.requests[streams[i]])
                {
                    auto frame_ref = archive->track_frame(streams[i]);
                    if (frame_ref)
                    {
                        //frame_ref->update_frame_callback_start_ts(std::chrono::high_resolution_clock::now());
                        //frame_ref->log_callback_start(capture_start_time);
                        on_before_requests(streams[i], frame_ref, archive);
                        (*config.requests[streams[i]])->on_frame(this, frame_ref);
                    }
                }
                else
                {
                    // Commit the frame to the archive
                    archive->commit_frame(streams[i]);
}

Following code additions  in zr300.cpp
void zr300_camera::on_before_requests(rs_stream stream, rs_frame_ref * frame, std::shared_ptr<rsimpl::frame_archive> archive)
    {
        if (!to_add_frames || stream != RS_STREAM_FISHEYE)
            return;

        auto_exposure->add_frame(clone_frame(frame), archive);
}

jlgarcia75, 2017-10-06:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello,

Do you still need help with this issue?

Best regards,
Jesus
Intel Customer Support

melax, 2017-10-06:
Was able to make it work.   Thanks.

closed 2017-10-06 by melax

Jdiesel666, 2018-05-21:
Hi @melax , I ran into the same issue with the fisheye auto exposure. And wondering where in the device.cpp you added that piece of code to make it work? 

Thanks a lot!