realsenseai/librealsense #472: When do you plan to support new RealSense Series?

nlyubova, 2017-04-04:
Hello,
do you plan to support RealSense 400 Series mentioned in news : https://newsroom.intel.com/chip-shots/intel-announces-tools-realsense-technology-development/ and provide an open-source driver? 

Thank you,
Natalia

alfredanil, 2017-04-04:
and more importantly when will R 400 be available for preorder? Also looking forward for intel euclid availability

dorodnic, 2017-04-04:
librealsense version supporting RS4xx devices is planned to be published to open-source as soon as the first of RS4xx models becomes available (I don't have concrete information regarding availability of the products themselves). 
We are taking the opportunity to revisit some of the design decisions with this release, so it will not be 100% API compatible with existing librealsense code, but will be close.

nlyubova, 2017-04-04:
sounds good! I'm looking forward for the new sensor to be available.
Do you also plan to support ROS wrapper for future sensors?

dorodnic, 2017-04-05:
Yes, ROS wrapper is also planned to be provided.

nlyubova, 2017-04-12:
Great, and thank you!

closed 2017-04-12 by nlyubova