realsenseai/librealsense #473: "There are 0 connected RealSense devices" message

eduard626, 2017-04-04:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0-71-generic |
| Build System |  Makefile |

eduard626, 2017-04-04:
I followed the installation instructions without major issues, I only had to update my kernel. After patching the uvcvideo kernel module I tried to run some example like "cpp-tutorial-1-depth", but I get the message saying 0 connected devices. When I plug the camera I cannot see it under /dev/video* . Has anyone faced this issue?

dorodnic, 2017-04-05:
Could you please run `dmesg -wT` when the device is unplugged, plug it in and share the output? 
This can determine if your Linux machine recognizes the connection of USB3 device or the problem is on a lower level.

eduard626, 2017-04-05:
The output of dmesg is

[Wed Apr  5 10:46:12 2017] usb 3-6: USB disconnect, device number 15
[Wed Apr  5 10:46:12 2017] usb 3-6.2: USB disconnect, device number 16

[Wed Apr  5 10:46:14 2017] usb 3-5: new high-speed USB device number 17 using xhci_hcd
[Wed Apr  5 10:46:14 2017] usb 3-5: New USB device found, idVendor=0451, idProduct=8027
[Wed Apr  5 10:46:14 2017] usb 3-5: New USB device strings: Mfr=0, Product=0, SerialNumber=1
[Wed Apr  5 10:46:14 2017] usb 3-5: SerialNumber: 4702186160B7
[Wed Apr  5 10:46:14 2017] hub 3-5:1.0: USB hub found
[Wed Apr  5 10:46:14 2017] hub 3-5:1.0: 2 ports detected
[Wed Apr  5 10:46:14 2017] usb 3-5.2: new high-speed USB device number 18 using xhci_hcd
[Wed Apr  5 10:46:14 2017] usb 3-5.2: New USB device found, idVendor=04b4, idProduct=22c3
[Wed Apr  5 10:46:14 2017] usb 3-5.2: New USB device strings: Mfr=1, Product=6, SerialNumber=0
[Wed Apr  5 10:46:14 2017] usb 3-5.2: Product: CX3-WinUsb
[Wed Apr  5 10:46:14 2017] usb 3-5.2: Manufacturer: Cypress

eduard626, 2017-04-06:
It turned out it was my USB port, I went through all my USB ports, only one of them worked.

closed 2017-04-06 by eduard626

Pierluigi1990, 2017-05-02:
Hi,
I have the same eduard626 issue. I tried to change USB port but nothing changed. Could anyone help me?
Thanks.

tony497, 2018-03-20:
Same issue too with the R200.
I got this message : 
[Di Mär 20 14:02:45 2018] usb 4-1.2-port3: Cannot enable. Maybe the USB cable is bad?
when i plug the camera and dmesg -wT is running. Any idea on how to deal with it ?

freemanlo, 2018-03-20:
R200 only supports USB3. Need to make sure if your USB connection is fine. Use "dmesg -wT" to see if it can be detected/recognized as a Super-Speed device. More of time, related to 1) the cable issue, 2) USB port or USB Hub capability.
Finding a USB3 qualified cable is a good approach.

tony497, 2018-03-20:
Thanks it work now.
I changed my cable and used it on a different usb port (also didn't worked with the docker station).