realsenseai/librealsense #479: rs_deproject_pixel_to_point()  elucidation

andakkino, 2017-04-20:
Hello GoodEvening to everybody,
I'm looking at the source code of  rs_deproject_pixel_to_point() function,

https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense/rsutil.h
line 43 in particular;

`float f = 1 + intrin->coeffs[0]*r2 + intrin->coeffs[1]*r2*r2 + intrin->coeffs[4]*r2*r2*r2;`

why coeffs[4]? It should be coeffs[2], in that formula it must consider the radial distortion coefficents instead. coeffs[4] is instead a tangential distortion coefficent.

Am I wrong? Can anyone help me to understand this function (and in particular the used formula) ?

thanks for your help.

delmottea, 2017-04-26:
the distortion equation is the same as used in openCV (and the coefficient order is the same) : 
http://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

the order is k1, k2, p1, p2, k3. It's standard order that allow to use the vector from one library to the other, and I think the reason is that it s ordered by importance k1 > k2 > p1,p2 > k3

leonidk, 2017-04-26:
It's not the same as opencv. Notice the difference in how tangential distortion is computed

delmottea, 2017-04-27:
oh, I didn't see it, thanks.

closed 2017-06-06 by dorodnic