realsenseai/librealsense #481: no matching media type for  pixel format 59555932

hinmanj, 2017-04-25:
| Required Info  |   |
|---|---|
| Camera Model | R200 (and SR300) | 
| Firmware Version |  1.0.72.10 | 
| Operating System & Version | Windows 10  |
| Build System |  VS2015 |

I'm getting this output as of late when trying to run the cpp_capture example upon startup (and any of my old projects utilizing the librealsense):

RealSense error calling rs_start_source(device:00000238A92AE9D0, source:VIDEO):
    no matching media type for  pixel format 59555932

Nothing has changed except for obvious Windows 10 updates between now and the last time I used the camera. Any ideas?

ev-mp, 2017-04-25:
The reported format 0x59555932 is (HEX for) 'YUY2'.

If the error is persistent with both your code and all the librealsense demos (did you try cpp-config-ui ?)
and additionally, the same camera worked correctly some time ago then I'd suggest to verify HW with Microsoft (3rd party) tool called GraphEdit:

[GraphEdit](https://msdn.microsoft.com/en-us/library/windows/desktop/dd407274%28v=vs.85%29.aspx?f=255&MSPPError=-2147217396) allows to configure and stream standard UVC color data from R200. It can also usable to verify Resolutions/FPS modes supported.
[image]
Please check and update

hinmanj, 2017-04-25:
This is what I'm seeing when trying to connect those pins.

I don't know anything about GraphEdit, but I did notice the "Video Camera Terminal" text atop your filter there doesn't exist for my RGB filter, but does for the Depth and Left/Right.

[image]

hinmanj, 2017-04-25:
I've noticed OBS is better for messing around with Directshow properties and such, and since I noticed this weird texture when trying to Render in GraphEdit *and* in OBS, this is what I see when trying to see RGB:

[image]

hinmanj, 2017-04-25:
Ok, after having tried updating drivers, restarting a lot, uninstalling and reinstalling the DCM, updating my graphics card drivers (I had installed a second GPU and figured it might have messed up my Directshow drivers since it seemed to be at a Directshow level), I finally just tried uninstalling only the RGB device in my device manager.

[image]

That worked for some reason. Even though I only tried uninstalling the R200 RGB, it seemed to fix my SR300 RGB as well, so... black magic.

quachconghoang, 2017-04-26:
Thanks @hinmanj !!!

closed 2017-04-26 by hinmanj

Maghoumi, 2017-06-09:
I faced this exact same problem after I updated my machine to Windows 10 with Creator's Update. Solution by @hinmanj worked for me as well.

MatsumotoJ, 2017-06-15:
I faced the same problem. However the solution by @hinmanj did not work for me. Specifically, the demos did not work without the driver. Also after reconnecting R200 or scanning HW change in the device manager the same version of the driver re-installed.

Could you give me any idea to fix the problem?

MatsumotoJ, 2017-06-15:
The solution by @hinmanj works now for me.
I updated Windows again after Windows Creator's Update.

Praattzz, 2017-07-03:
I am facing the same issue. I am trying to run the default librealsense demo (cpp-config-ui ) with R200 in Windows 10 with VS2017. I have issue with RGB camera. I noticed similar texture like @hinmanj with OBS. Following hinmanj steps - reinstalling DCM, updating GPU and rebooting system several times helped me. Now all three R200 capture cameras are looking good on OBS. Yet I am getting the same error "no matching media type for pixel format 59555932" when I run on VS. 

I updated Windows like @MatsumotoJ  tried and yet no luck. 
How to fix this problem ? Any help is highly appreciated.

MatsumotoJ, 2017-07-06:
I have no idea but

MatsumotoJ, 2017-07-06:
Hi Praattzz,

I have no idea but the problem happens every PC with the Creator's Update around me. Although the solution above worked in my case, it is a bit inconvenient.

Praattzz, 2017-08-24:
This issue is open now in Realsense Forums. The link to the thread is below:
 
https://communities.intel.com/thread/117338

P.S: If anyone found a way to fix it apart from the above methods, please let other know the logic too.

Thanks in advance.