realsenseai/librealsense #482: Turn on/off the infrared pattern

delmottea, 2017-04-26:
Hello,
I try to use multiple R200 (and later SR300) to get point cloud of an object from multiple viewpoints (around 6 cameras).
I discovered that the depth quality is worse when 2 cameras view the same face at the same time (interference of the pattern) so I'd like to turn on and off the IR projector to use only one camera at a time.
I don't need high framerate, but would need to get my 6 views in about 1 second.

I tried to stop the streaming and starting again, but it takes around 1 second per camera so too slow for my context.
Is there possibility to do the on/off faster in software or should I create an external shutter to occlude the projector?

thanks,
Arnaud

dorodnic, 2017-04-26:
Hello @delmottea 
For the R200 you can toggle `r200_emitter_enabled` option on and off, for SR300 you need `f200_laser_power`:
```cpp
dev.set_option(rs::option::r200_emitter_enabled, 0); //disable pattern
for (int i = 0; i < 10; i++) dev.wait_for_frames(); // wait several frames for the effect to take place
// get frame data
dev.set_option(rs::option::r200_emitter_enabled, 0); //re-enable pattern
```
Please note that there is certain latency between when you set option and when you get the first frame reflecting that change.

delmottea, 2017-04-27:
oh, perfect :)
Thank you very much.

It's not possible to get information if it was active on a specific frame or not?

RoboEvangelist, 2017-05-09:
How do you do this on SR300?

dorodnic, 2017-05-11:
@RoboEvangelist - For the SR300 there is `rs::option::f200_laser_power`, you can toggle it between 0 and 16 to switch laser off / full power. The purpose of the laser on the SR300 is different from R200, so when the laser is off you will not get any depth at all. 
@delmottea - Not out of the box... For the R200, there is a special "dinghy" data-structure that contains some information about each frame, but not the emitter state as far as I know. For the SR300, this information is available in the UVC metadata. Unfortunately, getting to it on Windows is very cumbersome (and only possible with Windows 10). On Linux, it is not exposed naively through V4L2. You can either hack it with custom patch to `uvcvideo` or wait until this feature gets up-streamed.

closed 2017-05-11 by dorodnic