realsenseai/librealsense #486: uvcvideo module installation error

mkotlyar, 2017-05-08:
Preparing Video4Linux backend as per https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation. 

Output from the ` ./scripts/patch-realsense-ubuntu-xenial.sh` script:
```
#
# configuration written to .config
#
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
Compiling uvc module
make: Entering directory '/home/tria/librealsense/ubuntu-xenial'
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_driver.o
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_queue.o
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_v4l2.o
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_video.o
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_ctrl.o
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_status.o
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_isight.o
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_debugfs.o
  CC [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_entity.o
  LD [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "media_entity_cleanup" [/home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_init" [/home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_create_link" [/home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
  LD [M]  /home/tria/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko
make: Leaving directory '/home/tria/librealsense/ubuntu-xenial'
Patched kernel module created successfully

Unloading module uvcvideo
Applying patched module uvcvideo
modprobe: ERROR: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
./scripts/patch-utils.sh: line 45: [: backup_available: integer expression expected
```

Output from `dmesg`:
```
[  550.076947] media: Linux media interface: v0.10
[  550.096834] Linux video capture interface: v2.00
[  550.122295] uvcvideo: loading out-of-tree module taints kernel.
[  550.122608] uvcvideo: module verification failed: signature and/or required key missing - tainting kernel
[  550.122952] uvcvideo: Unknown symbol media_entity_create_link (err 0)
[  550.123438] uvcvideo: Unknown symbol media_entity_init (err 0)
[  550.123736] uvcvideo: Unknown symbol media_entity_cleanup (err 0)
[  722.874901] uvcvideo: Unknown symbol media_entity_create_link (err 0)
[  722.875449] uvcvideo: Unknown symbol media_entity_init (err 0)
[  722.875748] uvcvideo: Unknown symbol media_entity_cleanup (err 0)
[  854.085205] uvcvideo: Unknown symbol media_entity_create_link (err 0)
[  854.085493] uvcvideo: Unknown symbol media_entity_init (err 0)
[  854.085633] uvcvideo: Unknown symbol media_entity_cleanup (err 0)
```

| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | unknown | 
| Operating System & Version | Ubuntu 16.04.1 |
| Kernel Version (Linux Only)  | 4.8.0-51-generic |
| Build System |  Makefile |

ev-mp, 2017-05-08:
It seem that you've upgraded to Ubuntu 16.04.2 with kernel 4.8.
The librealsense patches provided are for Ubuntu 16.04.1 and kernel 4.4. 
Basically you have a patched module compiled against sources of 4.4, which 4.8 kernel rejects  - hence the errors you receive.
Providing patches for 4.8 is scheduled, but at this point you need to switch back to original 16.04.1 kernel line of 4.4.0-xx for the script to succeed. 
This topic is also tracked [here](https://github.com/IntelRealSense/librealsense/issues/444)

jachu51, 2017-06-01:
Hi,

I used sources provided by _apt-get source_ instead of sources from the git repository and the script worked like a charm with kernel 4.8.0-53. I attach a modified script that uses sources from _apt-get source_. 
[patch-realsense-ubuntu-xenial.sh.zip](https://github.com/IntelRealSense/librealsense/files/1044483/patch-realsense-ubuntu-xenial.sh.zip)

Hope it helps!

JJHu1993, 2017-06-02:
I have modified the script to support new kernel(4.8.0-xx).
## link: https://github.com/JJHu1993/librealsense-patch.git

 ```
Patched kernel module created successfully

Unloading module uvcvideo
Applying patched module uvcvideo
Inserting uvcvideo succeeded

Script has completed successfully. Please consult the installation guide for further instruction.

jjhu@jjhu-computer:~/projects/librealsense$ uname -r
4.8.0-52-generic

```

@mkotlyar @ev-mp @jachu51

ev-mp, 2017-06-06:
Hi @JJHu1993 ,
Are you planning to send a pull request for 16.04.2/4.8 ? 
If not - I'm going to pick it up and incorporate into Ubuntu script so it will support both 4.4 and 4.8 kernel versions.

ev-mp, 2017-08-10:
We [updated kernel patch](https://github.com/IntelRealSense/librealsense/pull/569) to support the rolling kernel for Ubuntu 16.04.2 LTS ( promoted from 4.8 to 4.10). 
And now it is fully functional both with 16.04.1 and 16.04.2 distributions.

In case you're using Ubuntu 16.04.2 with kernel 4.8 - upgrade the system with "sudo apt update && sudo apt dist-upgrade" to get the upgrade.

**Important**   after switching from kernel 4.8 to 4.10 Librealsense kernel patch [should be reapplied](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#video4linux-backend-preparation).

The change does not effect Ubuntu 16.04.1 LTS users with kernel 4.4.0-....

A similar issue was reported and answered  [here](https://github.com/IntelRealSense/librealsense/issues/562)

closed 2017-08-10 by ev-mp