realsenseai/librealsense #489: R200 1080p synchronization issue

Luthien1990, 2017-05-09:
| Required Info  |    |
|---|---|
| Camera Model | R200 | 
| Firmware Version |  1.0.72.06   | 
| Operating System & Version |  Ubuntu  16.04 LTS |
| Kernel Version (Linux Only)  |  4.4.0-72 |
| Build System | Makefile |

I have been analysing the delay between frames of my R200 and I have found that the signal has a considerable amount of jitter. If the colour stream is set at 480p, this effect is delimited in an interval of (-1,1) ms but, with the 1080p colour configuration, the frame delay varies along a (-3,3) ms.

As this R200 device is being used along with other kind of depth sensors, in order to get a generic timestamp reference, I am using std::chrono::high_resolution_clock timestamps just after getting a new frame with wait_for_frames() method.

You can found one example of these results below.

1. When the camera is configured at:

- Color: 640x480@30fps bgr8 format

- Depth: 640x480@30fps z16 format

the difference between frame timestamps plot looks like this:
[image]

2. If the camera configuration is changed to: 

- Color: 1920x1080@30fps bgr8 format

- Depth: 640x480@30fps z16 format

the plot looks like that:
[image]

As it can be seen, in the 480p configuration the timestamp values are much more synchronized than in the 1080p case. Could it be happening because of an excessive computational burden in the device?

freemanlo, 2017-09-18:
Hello,
1) can you upgrade the FW to 1.0.72.10?
2) try below code to get the timestamp from R200

    // Set callbacks prior to calling start()
    auto depth_callback = [depth_intrin, depth_format](rs::frame f)
    {
        std::cout << depth_intrin.width << "x" << depth_intrin.height
            << " " << depth_format << "\tat t = " << f.get_timestamp() << "\tms" << std::endl;
    };
    auto color_callback = [color_intrin, color_format](rs::frame f)
    {
        std::cout << color_intrin.width << "x" << color_intrin.height
            << " " << color_format << "\tat t = " << f.get_timestamp() << "\tms" << std::endl;
    };

jlgarcia75, 2017-10-03:
For best support, post issues in the [RealSense Community](https://communities.intel.com/community/tech/realsense).

Hello @Luthien1990,

Do you still need help with this issue? If we do not hear from you in 7 days we will close this issue.

Best regards,
Jesus
Intel Customer Support

closed 2017-10-16 by jlgarcia75