realsenseai/librealsense #491: How to use librealsense on windows?

dan0910, 2017-05-10:
| Required Info  |   |
|---|---|
| Camera Model |   SR300 
| Firmware Version |   | 
| Operating System & Version |   Win10-64bit
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013,

Hello:
 
     I download the librealsense from github. But I can't find the **detail** installation direction about using librealsense on windows operation system.
    And My environment is Win10-64bit, VS2013, and realsense SR300.
 
   When I download the librealsense package, I open realsense.sln in "librealsense-master/librealsense.vc12" using VS2013. Then I build the project.
   But it comes two LINK  errors:
           error LINK1104: can't open the file realsense-sd.lib
           error LINK1104: can't open the file realsense-d.lib.
 
   I don't know how to slove it.
  Can you give me some suggestions?
  Thank you very much.

dorodnic, 2017-05-11:
Could you please provide the full listing from the Output Window?

dan0910, 2017-05-12:
Oh, sorry, there are  full error lists:
### 1. build"realsense-s"project in ".\librealsense-master\librealsense.vc12":
Error	1	error C1001: An internal error has occurred in the compiler.	e:\librealsense-master\src\archive.h	36	1	realsense-s

### 2. build "realsense"project in".\librealsense-master\librealsense.vc12":
Error	1	error C1001: An internal error has occurred in the compiler.	e:\librealsense-master\src\archive.h	36	1	realsense-s
Error	2	error LNK1104: cannot open file 'E:\librealsense-master\librealsense.vc12\realsense-s\obj\Debug-Win32\realsense-sd.lib'	E:\librealsense-master\librealsense.vc12\realsense\LINK	realsense

### 3. build an example project, such as"cpp-alignimages":
 Error	1	error C1001: An internal error has occurred in the compiler.	e:\librealsense-master\src\archive.h	36	1	realsense-s
Error	2	error LNK1104: cannot open file 'E:\librealsense-master\librealsense.vc12\realsense-s\obj\Debug-Win32\realsense-sd.lib'	E:\librealsense-master\librealsense.vc12\realsense\LINK	realsense
Error	3	error LNK1104: cannot open file 'realsense-d.lib'	E:\librealsense-master\librealsense.vc12\cpp-alignimages\LINK	cpp-alignimages

I find that the first error points to archive.h line 36 column 1, that is:   std::chrono::high_resolution_clock::time_point frame_callback_started {};

By the way, my VS 2013 version is: 12.0.30501.0 REL. I don't know if it's VS2013's fault...
I'm so sorry that I'm a rookie.
Thank you very much! (●'◡'●)

dorodnic, 2017-05-18:
This is probably related to the Visual Studio version. In theory we try to protect against it with a better error message, but the compiler probably never reached that part: 
```cpp
#if (_MSC_FULL_VER < 180031101)
    #error At least Visual Studio 2013 Update 4 is required to compile this backend
#endif
```
Can you please retry with Update 4?

dan0910, 2017-05-19:
Yes! I update VS2013 to update 5, and it works!
I appreciate your help so much !

dan0910, 2017-05-24:
resolved

closed 2017-05-24 by dan0910