realsenseai/librealsense #494: ZR300 crashes at startup when plugged in before booting computer

andrepfr, 2017-05-11:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Camera Firmware Version | 2.0.71.28  | 
| Operating System & Version |  Ubuntu 16.04.2 |
| Kernel Version (Linux Only)  |  4.4.0-77-generic  |
| Build System |  Makefile | 

In case the sensor is plugged into the computer before the computer boots up, the sensor crashes with the following error when launched:

```
[ INFO] [1494502600.596998307]: Initializing nodelet with 8 worker threads.
[ERROR] [1494502601.036582722]: /camera/driver - Error calling rs_get_device_info ( device:0xadce80, info:ADAPTER_BOARD_FIRMWARE_VERSION ): 
selected camera info is not supported for this camera! 

[FATAL] [1494502601.178728447]: Failed to load nodelet '/camera/driver` of type `realsense_camera/ZR300Nodelet` to manager `nodelet_manager'
[camera/nodelet_manager-2] process has died.
```

Note that it does not happen after a restart but only after a complete shutdown.
Any ideas on how to fix this?

helenol, 2017-05-18:
We have the same problem on Ubuntu 14.04.
This is blocking for us, as our sensor is mounted on a robot that we need to frequently shut down/restart.

ZacharyTaylor, 2017-05-18:
With this issue I have tried running as a super user, restarting uvcvideo, several command line options to reset usb devices and trying the `force_firmware_reset` and `force_hardware_reset` functions in librealsense. Nothing short of physically unplugging the camera and replugging it makes any difference.

skohlbr, 2017-05-23:
We had this issue with the R200 on 14.04 and the workaround was indeed re-plugging the camera after robot bootup (which was possible as the cable was exposed). Didn't find a solution on 14.04, but after switching to 16.04 the problem disappeared.

zubinp, 2017-07-10:
Did anyone find a solution yet? I have my sensor enclosed in my robot in such a way that I cannot unplug /replug it without opening multiple screws. Really need a software workaround!

zubinp, 2017-07-10:
I did find a temporary "hack" to atleast get the Rgb, IR and depth data. You will have to let go of the Fisheye and IMU data with this hack.  Read my answer in this thread: https://communities.intel.com/thread/116175

chinitaberrio, 2017-07-20:
Take a look at this:  https://github.com/ethz-asl/realsense

christiansprecher, 2017-08-25:
I experienced the same problem with the ZR300 and Ubuntu 16.04. The RealSense was thought to run on a closed system with frequent shutdowns. Constant unplugging and replugging is simply no option. This bug basically prevents me from using this camera. I hope this issue can be fixed from the side of Intel asap.

jlgarcia75, 2017-10-03:
Hello,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75

ZacharyTaylor, 2017-10-03:
Correct me if I am wrong but the D400 series lack a fisheye camera and IMU. This means that while they can replace the R200 they will typically be unsuited for any application where the additional capabilities of the ZR300 were needed. Is there no other alternative or further support?

jlgarcia75, 2017-10-03:
Hello @ZacharyTaylor,

Unfortunately, the ZR300 has been discontinued, so bugs are no longer being worked on. The focus is now on the D400 series cameras. If your issue is due to configuration or anything else that does not require a firmware, hardware, or software change, then we can try to help you. Otherwise, if your camera has issues that require a change in firmware, hardware, or software, then we cannot support it. If your camera is not functional and it is still under warranty, you can seek a refund from the point of purchase.

Regards,
Jesus G.
Intel Customer Support