realsenseai/librealsense #495: The color images acquired on Windows are somewhat reddish

chengricky, 2017-05-13:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |   | 
| Operating System & Version | Windows 10  |
| Build System |  VS2015 |

dorodnic, 2017-05-14:
Could you please provide more information? Reddish compared to Linux with the same camera?
Can you upload a picture? (You can drag and drop images into GitHub comment interface)

chengricky, 2017-05-14:
[image]
The same camera runing on Linux gets normal color imagea, as the upper picture.
But on windows, the color is reddish, as the below picture.
[image]

chengricky, 2017-05-14:
Addition: 
Another LR200 camera also acquires reddish color picture on Windows platform (VS2015).
However, camera gets normal pictures through Realsense SDK.

ddiakopoulos, 2017-05-14:
@chengricky this is not a bug in librealsense, per-say. The Intel RealSense SDK has a function somewhere that sets various color/hue/brightness/exposure presets before it starts streaming. I believe the ROS node for librealsense does something similar on Linux.

Please use the cpp-config-ui utility to experiment with color camera settings. Once you find reasonable values, you can programmatically set them in your own application as defaults.

dan0910, 2017-06-20:
@chengricky Hi，
Do you solve the reddish problem??
I also have this puzzle...
Do you use  cpp-config-ui utility to modify color camera settings?

chengricky, 2017-06-20:
@dan0910 no, i have not solved the problem.
I tried to modify some default camera parameters, such as white balance, contrast and saturation etc.
but it did not take effect at all.

jlgarcia75, 2017-10-03:
Hello @chengricky,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75