realsenseai/librealsense #496: OS X can't get IR and depth stream

AlexMa011, 2017-05-13:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version | OS X 10.12.4  |
| Kernel Version (Linux Only)  |   |
| Build System |   Makefile | 

I used brew to install librealsense and test the codes as in the opencv tutorial. I can run the color stream code but can't run the IR one. 

The error message prints like, "terminating with uncaught exception of type rs::error: Timeout waiting for frames."

dorodnic, 2017-05-14:
Hi,
Did you try to run included examples, like `cpp-config-ui`?

AlexMa011, 2017-05-22:
@dorodnic I tried it and I got error messages like these:

RealSense error calling rs_set_device_option(device:0x7fd2e0801620, option:COLOR_BACKLIGHT_COMPENSATION, value:1):
    libusb_control_transfer(...) returned LIBUSB_ERROR_NO_DEVICE

thunghan, 2017-06-09:
Hi, I have the same error, and can only run the color stream

viniciolindo, 2017-09-29:
hi same problem for me
high Sierra brew installation 1.12.1 version

jlgarcia75, 2017-10-17:
Hello,
This may be a problem with the USB ports in your computer. Did you try connecting the camera to a different computer?

Let us know if you still need help with this issue.

Regards,
Jesus G.
Intel Customer Support

closed 2017-10-26 by jlgarcia75