realsenseai/librealsense #497: Can I use librealsense under Windows 7 x64?

nizihabi, 2017-05-15:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version |   | 
| Operating System & Version |  Windows 7 x64 |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2013 | 

when I used this to initial sr300 under the Win7, it got "Unable to open device via WinUSB" error, does it support Windows 7?

dorodnic, 2017-05-18:
Hi,
librealsense does not support Windows 7 out of the box. The reason you are getting the error is because SR300 relies on a standard WinUSB driver. On later versions you see it as:
[image]
However, Win7 does not recognize this device. It is not impossible to get it to show-up, and then librealsense will most likely work fine. Google enabling WinUSB on Windows 7. 
That said, Windows 7 has several severe bugs inside Media Foundation core that Microsoft is not planning to patch, so proceed with caution.

nizihabi, 2017-05-20:
ok , thanks a lot : )

closed 2017-05-20 by nizihabi