realsenseai/librealsense #498: [Question] Is there have any plan to distribute this package to debian-source?

BlazingForests, 2017-05-16:
Hi, guys

Is there have any plan to distribute this package to debian-source?

I don't want to install ROS, so i don't want add ros-source list to my debian system.

But i want to install librealsense with KDMS. (like ros-kinetic-librealsense)

So, how can i do?

THX

BlazingForests, 2017-05-16:
BTW, I am using debian 9.0 (stretch) test source list

oordan, 2017-07-31:
This is our other option for librealsense:
```sh
$ echo 'deb "http://realsense-alm-public.s3.amazonaws.com/apt-repo" xenial main' | sudo tee /etc/apt/sources.list.d/realsense-latest.list
$ sudo apt-key adv --keyserver keys.gnupg.net --recv-key D6FB2970
$ sudo apt update 
```
(from [software.intel.com](https://software.intel.com/sites/products/realsense/intro/getting_started.html))

closed 2017-07-31 by oordan