realsenseai/librealsense #499: Kernel Issue With Ubuntu 16.04's New Kernel From Week of May 8th, 2017

RoboEvangelist, 2017-05-16:
| Required Info  |   |
|---|---|
| Camera Model | SR300 |       three of these cameras 
| Firmware Version |  3.21.0.0 (two sensors) and 3.15.0.0 (one sensor)  | 
| Operating System & Version |  Ubuntu 16.04 |
| Kernel Version (Linux Only)  |  4.8.0-51-generic |
| Build System | CMake 2.8+ | 

I upgraded my Kernel (on two different computers) last week, and now I have this issue where the realsense script complains about not having the right kernel version. I have the latest pull from realsense master branch. This is the message I get:
`modprobe: Error: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)
Failed to insert patched module. Operation is aborted, the original module restored
Verify that the current kernel version is aligned to the patched module version`

JJHu1993, 2017-06-02:
I have modified the script to support new kernel(4.8.0-xx).
## link: https://github.com/JJHu1993/librealsense-patch.git

 ```
Patched kernel module created successfully

Unloading module uvcvideo
Applying patched module uvcvideo
Inserting uvcvideo succeeded

Script has completed successfully. Please consult the installation guide for further instruction.

jjhu@jjhu-computer:~/projects/librealsense$ uname -r
4.8.0-52-generic

```

@RoboEvangelist

RoboEvangelist, 2017-06-02:
Aight. Imma try it, and then close the issue when successful.
Thanks!

ev-mp, 2017-06-21:
The patch has been merged into master branch and now both 4.4 and 4.8 LTS Ubuntu kernels are supported

closed 2017-06-21 by RoboEvangelist