realsenseai/librealsense #503: modprobe: ERROR: could not insert 'uvcvideo'

lixueyi83, 2017-05-19:
The below issue occurs when i follow the steps to compile uvc module. Could you please give me a hand what's going there? I am using Ubuntu16.04, with kernel version .8.0-52-generic.

=============================================================================
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release

  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
Compiling uvc module
make: Entering directory '/home/fts/Downloads/librealsense-master/ubuntu-xenial'
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_driver.o
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_queue.o
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_v4l2.o
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_video.o
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_ctrl.o
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_status.o
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_isight.o
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_debugfs.o
  CC [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvc_entity.o
  LD [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "media_entity_cleanup" [/home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_init" [/home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
WARNING: "media_entity_create_link" [/home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko] undefined!
  LD [M]  /home/fts/Downloads/librealsense-master/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko
make: Leaving directory '/home/fts/Downloads/librealsense-master/ubuntu-xenial'
Patched kernel module created successfully

Unloading module uvcvideo
Applying patched module uvcvideo
modprobe: ERROR: could not insert 'uvcvideo': Unknown symbol in module, or unknown parameter (see dmesg)
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version
./scripts/patch-utils.sh: line 45: [: backup_available: integer expression expected
==============================================================================

| Required Info  |   |
[image]

|---|---|
| Camera Model | R200 | 
| Firmware Version |   | 
| Operating System & Version |  Ubuntu 16.04|
| Kernel Version (Linux Only)  |  4.8.0-52-generic |
| Build System |  VS2013, XCode or Makefile |

JJHu1993, 2017-06-02:
replace librealsense/scripts/patch-realsense-ubuntu-xenial.sh to 
## https://github.com/JJHu1993/librealsense-patch.git
and run it again.

## Result:
```
Patched kernel module created successfully

Unloading module uvcvideo
Applying patched module uvcvideo
Inserting uvcvideo succeeded

Script has completed successfully. Please consult the installation guide for further instruction.

jjhu@jjhu-computer:~/projects/librealsense$ uname -r
4.8.0-52-generic

```

@lixueyi83

oordan, 2017-07-31:
#546

closed 2017-07-31 by oordan