realsenseai/librealsense #504: ZR300: Unable to get depth/ir data

zubinp, 2017-05-19:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |  2.0.71.28 | 
| Operating System & Version | Ubuntu Xenial 16.04.02  |
| Kernel Version (Linux Only)  |  4.4.0-78-generic |
| Build System | Makefile | 

I compiled and installed librealsense on my laptop using these instructions: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md    
The installation steps went fine without any errors. The kernel module patch was also successful. On attaching the ZR300 to my laptop and running cpp-capture , I can view RGB and Fisheye streams,both at 60fps, but I do not get any IR/depth data. The terminal from which I ran the cpp-capture command is flooded with the following error
```
...
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
rs.warn: Subdevice 1 bad magic number 0x0
rs.warn: Subdevice 0 bad magic number 0x0
...
```
And sometimes, these 2 streams freeze too, and I'm unable to restart cpp-capture after that.

I have tried this with 2 different ZR300 cameras and received same issues. I have tried patching the kernel module multiple times, and it is successful every time. Here is the terminal output:
```
sudo ./scripts/patch-realsense-ubuntu-xenial.sh
[sudo] password for sig: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.1ubuntu2).
git is already the newest version (1:2.7.4-0ubuntu1.1).
linux-headers-generic is already the newest version (4.4.0.78.84).
The following packages were automatically installed and are no longer required:
  dkms ros-kinetic-rgbd-launch ubuntu-core-launcher
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

First, remove all RealSense cameras attached. Hit any key when ready

Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
The kernel has modified files:
	modified:   drivers/media/usb/uvc/uvc_driver.c
	modified:   drivers/media/usb/uvc/uvcvideo.h
	modified:   include/uapi/linux/videodev2.h
Proceeding will reset all local kernel changes. Press 'n' within 10 seconds to abort the procedure
Do you want to proceed? [Y/n]y

Resetting local changes in ubuntu-xenial folder
 HEAD is now at 156e000 UBUNTU: Ubuntu-4.4.0-78.99
Update the folder content with the latest from mainline branch
From git://kernel.ubuntu.com/ubuntu/ubuntu-xenial
 * branch            master     -> FETCH_HEAD
Already up-to-date.
Applying F200 formats patch patch
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file include/uapi/linux/videodev2.h
Applying ZR300 SR300 and LR200 formats patch
patching file drivers/media/usb/uvc/uvc_driver.c
patching file drivers/media/usb/uvc/uvcvideo.h
patching file include/uapi/linux/videodev2.h
scripts/kconfig/conf  --silentoldconfig Kconfig
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CHK     include/generated/bounds.h
  CHK     include/generated/timeconst.h
  CHK     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
Compiling uvc module
make: Entering directory '/home/sig/librealsense/ubuntu-xenial'
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_driver.o
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_queue.o
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_v4l2.o
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_video.o
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_ctrl.o
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_status.o
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_isight.o
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_debugfs.o
  CC [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvc_entity.o
  LD [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.o
  Building modules, stage 2.
  MODPOST 1 modules
  LD [M]  /home/sig/librealsense/ubuntu-xenial/drivers/media/usb/uvc//uvcvideo.ko
make: Leaving directory '/home/sig/librealsense/ubuntu-xenial'
Patched kernel module created successfully

Unloading module uvcvideo
Applying patched module uvcvideo
Inserting uvcvideo succeeded

Script has completed successfully. Please consult the installation guide for further instruction.
```

 I am lost here now and could use some help. Thanks!

oordan, 2017-07-30:
Please run `scripts/gt_setup_verification.sh` to verify kernel patches status (seem to be Ok)
Also, please submit the output of `dmesg -T` when this issue happens. You can force verbose kernel logs by issuing `sudo echo 0xFFFF > /sys/module/uvcvideo/parameters/trace` before running the application and calling dmesg

zubinp, 2017-08-04:
@oordan , thanks for replying. Unfortunately the laptop on which I was facing the above issue is no longer in my possession , else I would have gone ahead with your debugging instructions. I suspect that the issue was with the USB Driver firmware in that laptop (Thinkpad W510). I will close this for now and if some one faces any similar problems he can start a new issue.

closed 2017-08-04 by zubinp

MPolymath, 2017-08-17:
I had this same issue.
Following the **Video4Linux backend preparation** section of the tutorial below fixed the issue:

```
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
```

The uvc (USB video class) module needs to be patched.

Maxime59, 2017-09-16:
I think I have the same issue regarding the detection of the ZR300 and can't understand what's going on here...

The script `gt_setup_verification.sh` returns me this : 
```
Host Info:

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

Kernel Version:

Linux 4.4.0-89-generic x86_64

Connected UVC Devices: 

UVC streaming device was found...

ZR300 Camera was found - Ok

Verify formats required by ZR300 Camera: 

Y8I	-	Ok
Y12I	-	Ok
Y16	-	Ok
Z16	-	Ok
GREY	-	Ok
YUYV	-	Ok
RW16	-	Ok

The pixel formats are set properly! 

Done...
```

I can access to the RGB feed through cheese or guvcview (also what looks like the depth data) however I can't run any example because I have the following message : `No device detected. Is it plugged in?` or `There are 0 connected RealSense devices.` .

When I try to run my code which was running with SR300 I get the following error : 
`terminate called after throwing an instance of 'rs::error'
  what():  out of range value for argument "index"
Aborted (core dumped)`

Please someone help me

jlgarcia75, 2017-10-03:
Hello @Maxime59,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

daeunSong, 2017-12-22:
Hello @jlgarcia75 ,
I am having same issue here,

I bought my ZR300 last month, and I never got any notices that Intel will no longer be updating the software. I had no other camera choice back then that D400 series pre-order was not opened. 

I wonder what you mean by the **replacement**. 

THANX.