realsenseai/librealsense #505: Depth image flickering/jumping

nlyubova, 2017-05-23:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | 3.15.0.0 | 
| Operating System & Version | Ubuntu 16.04 |
| Kernel Version (Linux Only)  | 4.4.15-tegra  |
| Build System |  Makefile | 

Depth image is flickering/jumping and cut in the middle, looks like there is a problem with a buffer size?

[image]

JasonChang13, 2017-09-19:
hi @nlyubova , would like to know if you still have this issue on latest LibRealsense v2.7.9, rs-capture sample?

wwppyy, 2017-09-22:
Hi @nlyubova you could remove one image ex: remove IR image screen rendering to see if that help

as to why the depth image would more stable (assume what you mean is depth value more consistent over time) when more close to object, 
(1) the SR300 features a  laser projector to project invisible IR signal to object and take its refection to calculate the depth, so strength of reflection IR signal matter.
(2)  any external light (sunlight) has wavelength close to IR wavelength SR300 using will cause interference.

closed 2017-10-17 by jlgarcia75