realsenseai/librealsense #506: error: ‘GLFW_FLOATING’ was not declared

thomasck, 2017-05-23:
I am trying to install librealsense on my UP-board running ubilinux. During the make && make install command I get the error

[ 61%] Building CXX object examples/CMakeFiles/cpp-config-ui.dir/cpp-config-ui.cpp.o
/home/tck/librealsense/examples/cpp-config-ui.cpp: In function ‘void show_message(GLFWwindow*, const string&, const string&)’:
/home/tck/librealsense/examples/cpp-config-ui.cpp:272:20: error: ‘GLFW_FLOATING’ was not declared in this scope
     glfwWindowHint(GLFW_FLOATING, GL_TRUE);
                    ^
examples/CMakeFiles/cpp-config-ui.dir/build.make:54: recipe for target 'examples/CMakeFiles/cpp-config-ui.dir/cpp-config-ui.cpp.o' failed
make[2]: *** [examples/CMakeFiles/cpp-config-ui.dir/cpp-config-ui.cpp.o] Error 1
CMakeFiles/Makefile2:420: recipe for target 'examples/CMakeFiles/cpp-config-ui.dir/all' failed
make[1]: *** [examples/CMakeFiles/cpp-config-ui.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2

Can anyone help me with this?

Thanks,

Thomas

closed 2017-05-25 by thomasck

parkerowan, 2017-06-04:
Having the same error, how did you resolve this?

thomasck, 2017-06-04:
Before you run the make command, go into the scripts folder and run all the .sh scripts - that should fix it. Some of the scripts will fail, but the ones that run fix the problem.. i think. Not a great explanation but it worked for me.

parkerowan, 2017-06-04:
Thanks for the quick reply. It seems that the install_glfw3.sh added the missing dependency.

thomasck, 2017-06-04:
No problem, ah that makes sense.