realsenseai/librealsense #507: Exceptions thrown when creating context after an improper shutdown

MikeGameDev, 2017-05-24:
| Required Info  |   |
|---|---|
| Camera Model | F200 | 
| Firmware Version | 2.6  | 
| Operating System & Version | Windows 10 Build 14393  |
| Kernel Version (Linux Only)  |   |
| Build System |  VS2015 | 

When attempting to initialize the camera the following exceptions are sometimes thrown:

> calibration table is not compatible with this API
> opcodes do not match

Note that the exception thrown will alternate between the two, seemingly at random. So, sometimes when the program is run, the first exception is thrown, other times the second one gets thrown. The exceptions get thrown when creating the rs::context object.

This seems to happen when a program which creates a rs::context gets shut down in a strange way. Once these exceptions start occurring, the only way to regain control of the camera is to restart the computer.

When running the 'cpp-capture' example with the logging set to rs::log_severity::debug, this is what gets outputted:

> rs.info: UVC device detected with VID = 0x5a9 PID = 0xf580
> rs.info: UVC device detected with VID = 0x8086 PID = 0xa66
> RealSense error calling rs_create_context(api_version:11201):
>     calibration table is not compatible with this API

Any thoughts on how I can regain control of the camera, without needing to restart the computer?

Thanks!

dorodnic, 2017-05-29:
Could you please try adding:
```cpp
WinUsb_ResetPipe(usb_interface_handle, 0x81); WinUsb_ResetPipe(usb_interface_handle, 0x01);
```
to uvc-wmf.cpp, right after:
```cpp
if(!WinUsb_Initialize(usb_file_handle, &usb_interface_handle)) { LOG_ERROR("Last Error: " << GetLastError()); throw std::runtime_error("could not initialize winusb"); }
```
Let me know if this helps

MikeGameDev, 2017-05-29:
I have added those lines. They seem to cause the application to hang once it tries to read from the usb device. After waiting for a couple of minutes, I paused the execution and copied the following call stack. 
After removing the 'WinUsb_ResetPipe' calls and running the app again, the application still hung in the same spot. Restarting the computer and running the program with the 'WinUsb_ResetPipe' calls removed allowed the app to run as expected. 

The camera is built into a HP Sprout computer, so there is no way for me to do a simple unplug / replug.

>  	ntdll.dll!00007ffcd7d96154()	Unknown
 	KernelBase.dll!00007ffcd4333acf()	Unknown
 	KernelBase.dll!00007ffcd4325d6c()	Unknown
 	winusb.dll!00007ffcc2fc2a6d()	Unknown
**>	realsense-d.dll!rsimpl::uvc::device::usb_synchronous_read(unsigned char endpoint, void * buffer, int bufferLength, int * actual_length, unsigned long TimeOut) Line 657	C++ THIS IS WHERE IT HANGS**
 	realsense-d.dll!rsimpl::uvc::bulk_transfer(rsimpl::uvc::device & device, unsigned char endpoint, void * data, int length, int * actual_length, unsigned int timeout) Line 796	C++
 	realsense-d.dll!rsimpl::hw_monitor::execute_usb_command(rsimpl::uvc::device & device, std::timed_mutex & mutex, unsigned char * out, unsigned __int64 outSize, unsigned int & op, unsigned char * in, unsigned __int64 & inSize) Line 66	C++
 	realsense-d.dll!rsimpl::f200::get_f200_calibration_raw_data(rsimpl::uvc::device & device, std::timed_mutex & usbMutex, unsigned char * data, unsigned __int64 & bytesReturned) Line 359	C++
 	realsense-d.dll!rsimpl::f200::read_f200_calibration(rsimpl::uvc::device & device, std::timed_mutex & mutex) Line 430	C++
 	realsense-d.dll!rsimpl::make_f200_device(std::shared_ptr<rsimpl::uvc::device> device) Line 270	C++
 	realsense-d.dll!rs_context_base::rs_context_base() Line 72	C++
 	realsense-d.dll!rs_context_base::acquire_instance() Line 98	C++
 	realsense-d.dll!rs_create_context(int api_version, rs_error * * error) Line 102	C++
 	cpp-tutorial-2-streams-d.exe!rs::context::context() Line 321	C++
 	cpp-tutorial-2-streams-d.exe!main() Line 18	C++
 	cpp-tutorial-2-streams-d.exe!invoke_main() Line 65	C++
 	cpp-tutorial-2-streams-d.exe!__scrt_common_main_seh() Line 253	C++
 	cpp-tutorial-2-streams-d.exe!__scrt_common_main() Line 296	C++
 	cpp-tutorial-2-streams-d.exe!mainCRTStartup() Line 17	C++
 	kernel32.dll!00007ffcd6bc8364()	Unknown
 	ntdll.dll!00007ffcd7d570d1()	Unknown

MikeGameDev, 2017-06-12:
It seems that this problem isn't necessarily related to an unexpected app exit. In the last week I have been ensuring that the rs::context destructor gets called. Eventually the camera just stops working as described in the original post. It seems to be completely random. Disabling / enabling the camera from the device manager has no effect, forcing me to restart the PC.

MikeGameDev, 2017-06-12:
More info: sometimes a "mutex destroyed while busy" exception gets thrown on exit. Once this happens, subsequent camera initialization attempts fail. I haven't been able to figure out which mutex is having issues, will update once I have more info.

jlgarcia75, 2017-10-03:
Hello @MikeGameDev,

If the issue is software related, we are no longer updating this software as we have moved to our latest generation, the D400 series. If you are in your warranty, a replacement can be provided.

Jesus G.
Intel Customer Support

closed 2017-10-03 by jlgarcia75