realsenseai/librealsense #508: how to change the depth, for a particular environment !!

zubair123khan, 2017-05-24:
| Required Info  |   |
|---|---|
| Camera Model | R200 |  
| Operating System & Version | ubuntu 16.04  |
| Kernel Version (Linux Only)  |  4.4.0-78-generic |

hello guys, 
i have a new R200 camera and the depth i am seeing in cpp-capture is very bad , and when i change it to something better, it again goes back to same condition as before, 
please help me find a way to set the depth if the camera properly . 

thanks

zubair123khan, 2017-05-24:
[image]

this the screen shot of my bad depth

dorodnic, 2017-05-29:
As the first step I would recommend checking the value of `RS_OPTION_R200_EMITTER_ENABLED` option (should be 1) and `RS_OPTION_R200_LR_AUTO_EXPOSURE_ENABLED`. If auto-exposure is enabled, try disabling it and playing around with LR exposure and gain in `cpp-config-ui` until you reach optimal results.

szx0112, 2017-07-11:
Hi, I am interesting to know where to change these parameters. Directly changing parameters in cpp-configure-ui won't saved. Thanks.

mojovski, 2017-07-14:
I guess its inside the C/C++ API (library)

oordan, 2017-07-27:
There is no way to save the parameters persistently into the camera. Indeed, the way to restore values it through the application.

closed 2017-07-27 by oordan