realsenseai/librealsense #511: patch-arch.sh fails to patch videodev2.h

william01110111, 2017-05-25:
| Required Info  |   |
|---|---|
| Camera Model | N/A | 
| Firmware Version | N/A | 
| Operating System & Version | Arch Linux (updated May 24, 2017) |
| Kernel Version (Linux Only)  | 4.11.2-1 |
| Build System |  Makefile | 

In the kernel patching stage of the install process, I get this error:
```
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 627.
```

[image]

seer-it, 2017-07-05:
Yes it is the behaviour for this kernel and all new ones.
I changed two files : patch-arch.sh and the patch itself and it worked.
I can upload them for you.

However now with Kernel 4.11.7 i have new problems (it worked before that).

runjerry, 2017-07-13:
Hi I have got the same issue, would you share your modifications to the two files? Thanks!

seer-it, 2017-07-17:
Tell me if it works (works on my environment).
https://github.com/seer-it/librealsense/tree/arch-fix

william01110111, 2017-07-19:
It worked! thanks @seer-it. Can you get your fix merged back to the main repo?

seer-it, 2017-07-19:
Glad to hear that, I'll make pull request.

runjerry, 2017-07-21:
Thanks @seer-it! The patch and compile work well after modifying the patch-realsense-ubuntu-xenial.sh in a similar way as you did for patch-arch.sh. But I got some runtime problem though, which I think is related to my hardware, the RealSense SR300 seems to be very picky about voltage and USB port.

william01110111, 2017-07-21:
Same. With my MacBook Pro (running Arch Linux) the example software crashes if the RealSense is plugged into one USB port, but not the other. Does this sort of thing always happen or is it related to the Linux software?

seer-it, 2017-07-22:
From what we've tested in company It seems that SR300 has a power requirements that USB port can't handle (at times).  It is hardware problem which we've solved with buying powered USB hub.

You can check if camera driver crashes after plugging into usb with 
`dmesg`

xoryouyou, 2017-07-24:
@seer-it i've cloned you arch-fix branch and have run `scripts/patch-arch.sh` but I still get the error when patching arch.

```
RealSense patch...
patching file drivers/media/usb/uvc/Makefile
patching file drivers/media/usb/uvc/uvc_driver.c
Hunk #1 succeeded at 188 with fuzz 2 (offset 40 lines).
patching file drivers/media/usb/uvc/uvcvideo.h
Hunk #1 succeeded at 127 with fuzz 2 (offset 8 lines).
patching file include/uapi/linux/videodev2.h
Hunk #1 FAILED at 627.
1 out of 1 hunk FAILED -- saving rejects to file include/uapi/linux/videodev2.h.rej
```

Is there any patch I have to run beforehand ?

The Current Kernel is `Linux 4.11.9-1-ARCH`

runjerry, 2017-07-25:
Thanks @seer-it! It works on some ports on my desktop with a 3 feet cable, but not with a 15 feet one. I am going to order a powered USB hub for it.

oordan, 2017-07-26:
Thanks @seer-it. We'll be glad for a pull request...

jlgarcia75, 2017-10-05:
Closing issue as it seems this has been resolved.

closed 2017-10-05 by jlgarcia75