realsenseai/librealsense #513: Realsense R4xx series camera support

thexperiments, 2017-05-29:
Hi,
Any news on the release of librealsense for R4xx series cameras?
I have the hardware but no SDK at the moment. I already reached out to my contact at Intel but don't have a response yet.

Regards

| Required Info  |   |
|---|---|
| Camera Model | R410 | 
| Firmware Version |   | 
| Operating System & Version |   |
| Kernel Version (Linux Only)  |   |
| Build System |   |

nlyubova, 2017-05-30:
Do you have some information about when will it be available for purchase? at least approximate?

abahubal, 2017-09-12:
Hi, 
     The latest SDK 2.0 supporting RS4xx Series is available for Download on [Intel® RealSense™ SDK 2.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.7.7)

jlgarcia75, 2017-09-14:
Hi @nlyubova, the RealSense D400 series of cameras will be available for purchase in October: https://software.intel.com/en-us/realsense/400-series. The librealsense branch that supports it is not final yet but will be final at launch of the product.

Jesus
Intel Customer Support

nlyubova, 2017-09-19:
@jlgarcia75 thank you for the information! Looking forward to try it!

closed 2017-09-19 by jlgarcia75