realsenseai/librealsense #514: Is realsense-camera-formats.patch compatible with SR300?

nlyubova, 2017-05-30:
| Required Info  |   |
|---|---|
| Camera Model | SR300 | 
| Firmware Version | tested both 3.15.0.0 and 3.21.0.0 | 
| Operating System & Version | Ubuntu 16.04 |
| Kernel Version (Linux Only)  | 4.4.15-tegra  |
| Build System |  Makefile | 

Is realsense-camera-formats.patch compatible with SR300? This patch applies successfully for us on 4.4.15-tegra kernel but the depth image is corrupted: jumping, and cut in the middle (see the screenshot).

It seems that some formats from 0002-LR200-ZR300-and-SR300-Pixel-Formats.patch are different in realsense-camera-formats.patch. Please help to adapt realsense-camera-formats.patch to make it compatible with SR300 and 4.4.15-tegra kernel because this is the only patch that can be successfully applied on 4.4.15-tegra. 

[image]

ev-mp, 2017-06-06:
Hi, the patch is applied correctly, otherwise you wouldn't be able to see IR and Depth at all.
It seem that the frame is corrupted during transmit . You can see that the first several lines (in IR) image are not synced at all, the sync becomes stable, but occurs in the middle of the frame.
Is this a commercial SR300 or an engineering sample ? Did you try to use a different SR300 ?

nlyubova, 2017-06-06:
Thank you for your answer. We have tried two different SR300 (an older version and a just-bought version) and both give the same kind of corrupted depth image. However, both cameras work well on a PC (Ubuntu 14 and 16) so the problem occurs only for Jetson TX2. What can be a problem?

dorodnic, 2017-06-06:
This type of problems also occur frequently when working inside VM, or when CPU usage is high. `uvcvideo` kernel module is not keeping up and some of the URB packets are getting lost, resulting in a corrupt image. 
There is a way to increase the number of URBs in the kernel module, but this is not a patch we can recommend (on some systems it had unexpected side-effects, and this change can never be upstreamed). This change, however, can help with the issue, at expense of slightly higher latency and higher memory consumption.

nlyubova, 2017-06-13:
Thank you ! That makes sense.
Why does R200 and ZR300 work well on the same system ? with the same patch, same driver and the same USB port?

dorodnic, 2017-06-17:
@nlyubova, unfortunately I don't have an answer for that. Jetson TX2 is not a platform we have ever tested and we don't have access to it for now. SR00 does not share the same firmware and USB micro-controller as R200 & ZR300. What we would usually do is look into the USB transactions using USB analyzer, such as Lecroy. If you could send Wireshark USB traces, this could possibly also reveal useful information.

closed 2017-07-01 by dorodnic