realsenseai/librealsense #516: advice for ZR300 multicam setup / settings

da9e, 2017-05-31:
I apologize before hand if this is not the right place to post this kind of question.

I'm currently attempting to do something similar to a real-time 3D scanner using a multicam setup.
I bought the ZR300 realsense due to this description in the datasheet: "In general, the ZR300 is not subject to infrared projector interference from multiple cameras in use simultaneously." ([source](https://click.intel.com/media/ZR300-Product-Datasheet-Public-002.pdf))

Now I have been running a small test setup using the multicam-cpp example bundled with librealsense with 2xZR300 and a 1xSR300 connected. I put a small video up of what I found [here](https://www.youtube.com/watch?v=PVOFtoR_B58&feature=youtu.be). The two ZR300's are the left and middle windows, while the SR300 is the right-most windows. Looking at the depth image SR300 has a very nice clean depth image with almost no missing data (black pixels). However the two ZR300's have more noise and it becomes obvious (look at timestamp 0:35 to 1:40 in the video) that the areas which both IR projectors hit are blocking the depth measurements from for both ZR300 (creating black pixels). I find this a little odd thinking of the quote from the data sheet, but maybe I'm misreading "interference"?

I read in the data sheet that the active projector is only there to help in scenarios where texture is missing (e.g. measuring depth on a unicolored wall). Would it be possible to turn off the projectors and only rely on the stereo vision based depth? Do you have other suggestions for optimizations that I could do to improve on this multisensor setup?

 Thank you.

|---|---|
| Camera Model | ZR300 | 
| Firmware Version | No idea | 
| Operating System & Version | Windows 10 |
| Build System |  VS2017 Community |

dorodnic, 2017-06-06:
I am not an expert on the internal hardware algorithms but based on my experience we should try to troubleshoot this issue by modifying the application to display Infrared instead of Color. Looking directly at the infrared input can give us a clue about what is going on. I'm concerned that some of the infrared images are getting over or under saturated and the auto-exposure is either disabled or not doing its job. 
ZR300 camera is generating depth from stereo. This means that if the infrared image has a lot of details, the projector is optional. If the image is "flat", the projector is artificially injecting more details into the scene, and it is done in a way that should not be destructive. However, if the extra details are not visible in the infrared image (either they are influenced by SR300 patterns, or the image has low contrast or is very bright or dark) then you will only get proper depth on objects edges.

dorodnic, 2017-07-01:
Hi @da9e, please let us know if you have additional questions. 
You might also get more information at the [community forum](https://communities.intel.com/community/tech/realsense).

closed 2017-07-01 by dorodnic