realsenseai/librealsense #519: SR300 infrared projector doesn't switch off if too close object

delmottea, 2017-06-02:
Hello,
I have a problem to set the power of the infrared projector of my SR300 (firmware 3.15), I discovered that if an object is too close, the projector stays on even if I set to 0.
Here is a video of my experiment : https://www.youtube.com/watch?v=knnLdWL8SXg
I use a realsense SR300, taking only the color and infrared flux, depth disabled.
rsCamera->enable_stream(rs::stream::infrared, 640, 480, rs::format::y8, 30);
rsCamera->enable_stream(rs::stream::color, 1280, 720, rs::format::rgb8, 10);
The code just loop and variate the laser power with set_option(rs::option::f200_laser_power, power) and print the state and the power value on the screen (the state "off" means we set the value to 0)
I observed that if an object is too close (roughly below 60cm) to the camera at initialisation, the laser power stays fixed and doesn't change value until I liberate the area. After I removed the object, I can put back in place, everything will work fine, but I have no guarantee it will be the case in my scenario.

delmottea, 2017-06-06:
I tried this morning and this bug didn't happen anymore. Probably happen only after long use of the camera (overheat? Disconnecting and reconnecting the camera, restarting the computer didn't solved the problem last time).

oordan, 2017-07-27:
This is an unexpected behavior. Please reopen if the issue reproduces.

closed 2017-07-27 by oordan