realsenseai/librealsense #521: Is there a way to record and playback using R200?

chanhyeoni, 2017-06-05:
| Required Info  |   |
|---|---|
| Camera Model | **R200** | 
| Firmware Version |  **Not sure**  | 
| Operating System & Version | **Ubuntu 16.04**  |
| Kernel Version (Linux Only)  |  **Not Sure** |
| Build System |  **Makefile** | 

I have trouble finding the way for me to record and do the playback with R200. Any reference or things for me to look up is fine. Thank you.

dorodnic, 2017-06-06:
Hi @chanhyeoni 
Playback and record capability (into ROS-bag files) on top of librealsense are provided as part of [github.com/IntelRealSense/realsense_sdk](https://github.com/IntelRealSense/realsense_sdk). Please give it a try and reopen the issue if there is a problem.

closed 2017-06-06 by dorodnic

chanhyeoni, 2017-06-06:
Hi @dorodnic,
Thanks for reply. I am aware of the SDK, and I am concerned that it is only compatible with ZR300, not R200. I will try to use it for R200; if it doesn't work, I have to find other ways.

prachi220, 2018-02-18:
Hi @chanhyeoni ,
I am working on a project where I am required to record the videos from R200 and save it on my laptop. I wanted to know if you managed to figure out a way to record. Any help would be highly appreciated :)