realsenseai/librealsense #522: ZR300 Motion module not reporting

joemenke, 2017-06-06:
| Required Info  |   |
|---|---|
| Camera Model | ZR300 | 
| Firmware Version |   |
    CAMERA_FIRMWARE_VERSION: 	2.0.71.28
    ADAPTER_BOARD_FIRMWARE_VERSION: 	1.29.0.0
    MOTION_MODULE_FIRMWARE_VERSION: 	1.25.0.0 
| Operating System & Version |   |
OSX 10.11.6
| Build System |  Cmake/Makefile | 

The motion module on my sensor does not seem to be reporting correctly. I am not sure if this is a hardware problem or a librealsense bug. Running both cpp-motion-capture and cpp-config-ui examples there will be maybe one callback from the motion module and then nothing from then on. Has anyone had any experience with this?

joemenke, 2017-06-07:
It appears this is a OSX specific issue. Potentially related to libusb / usb port permissions. It runs correctly on Windows. 

Some more details:
The cameras run correctly, just the motion module stops reporting. cpp-enumerate-devices gives this output consistently:

    CAMERA_FIRMWARE_VERSION: 	2.0.71.28
    ADAPTER_BOARD_FIRMWARE_VERSION: 	1.29.0.0
    MOTION_MODULE_FIRMWARE_VERSION: 	1.25.0.0
    CAMERA_TYPE         : 	PRQ-Ready
    OEM_ID              : 	OEM None
    ISP_FW_VERSION      : 	0xd942
    CONTENT_VERSION     : 	12
    MODULE_VERSION      : 	4.4.0.111
    IMAGER_MODEL_NUMBER : 	33
    CALIBRATION_DATE    : 	2014-07-04 08:18:35 UTC
    EMITTER_TYPE        : 	Laser Driver 4.1
    FOCUS_VALUE         : 	0
    LENS_TYPE           : 	Newmax 58.9 x 45.9 degs in VGA
    3RD_LENS_TYPE       : 	AEOT
    LENS_COATING__TYPE  : 	Visible-light block / IR pass 43 nm width
    3RD_LENS_COATING_TYPE: 	IR coating
    NOMINAL_BASELINE    : 	70 mm
    3RD_NOMINAL_BASELINE: 	58 mm
 Supported options:                                    min        max       step  default  
    COLOR_BACKLIGHT_COMPENSATION                       : 0    ... 1           1     0         
    COLOR_BRIGHTNESS                                   : 65472... 64          1     0         
    COLOR_CONTRAST                                     : 0    ... 100         1     50        
    COLOR_EXPOSURE                                     : 50   ... 666         1     166       
    COLOR_GAIN                                         : 0    ... 128         1     64        
    COLOR_GAMMA                                        : 100  ... 500         1     300       
    COLOR_HUE                                          : 0    ... 0           1     300       
    COLOR_SATURATION                                   : 0    ... 100         1     64        
    COLOR_SHARPNESS                                    : 0    ... 100         1     50        
    COLOR_WHITE_BALANCE                                : 2800 ... 6500        10    4600      
    COLOR_ENABLE_AUTO_EXPOSURE                         : 0    ... 1           1     1         
    COLOR_ENABLE_AUTO_WHITE_BALANCE                    : 0    ... 1           1     1         
    R200_LR_AUTO_EXPOSURE_ENABLED                      : 0    ... 1           1     0         
    R200_LR_GAIN                                       : 100  ... 6399        1     400       
    R200_LR_EXPOSURE                                   : 1    ... 330         1     164       
    R200_EMITTER_ENABLED                               : 0    ... 1           1     0         
    R200_DEPTH_UNITS                                   : 0    ... 2.14748e+09 1     1000      
    R200_DEPTH_CLAMP_MIN                               : 0    ... 65535       1     0         
    R200_DEPTH_CLAMP_MAX                               : 0    ... 65535       1     65535     
    R200_DEPTH_CONTROL_ESTIMATE_MEDIAN_DECREMENT       : 0    ... 255         1     5         
    R200_DEPTH_CONTROL_ESTIMATE_MEDIAN_INCREMENT       : 0    ... 255         1     5         
    R200_DEPTH_CONTROL_MEDIAN_THRESHOLD                : 0    ... 1023        1     192       
    R200_DEPTH_CONTROL_SCORE_MINIMUM_THRESHOLD         : 0    ... 1023        1     1         
    R200_DEPTH_CONTROL_SCORE_MAXIMUM_THRESHOLD         : 0    ... 1023        1     512       
    R200_DEPTH_CONTROL_TEXTURE_COUNT_THRESHOLD         : 0    ... 31          1     6         
    R200_DEPTH_CONTROL_TEXTURE_DIFFERENCE_THRESHOLD    : 0    ... 1023        1     24        
    R200_DEPTH_CONTROL_SECOND_PEAK_THRESHOLD           : 0    ... 1023        1     27        
    R200_DEPTH_CONTROL_NEIGHBOR_THRESHOLD              : 0    ... 1023        1     7         
    R200_DEPTH_CONTROL_LR_THRESHOLD                    : 0    ... 2047        1     24        
    FISHEYE_EXPOSURE                                   : 2    ... 320         1     4         
rs.error: libusb_release_interface(...) returned LIBUSB_ERROR_NOT_FOUND
Segmentation fault: 11

dorodnic, 2017-06-07:
We are aware of the fact that ZR300 is not functioning perfectly on OSX platform. My personal experience on OSX10 was that the camera was performing streaming, and motion tracking, but not together at the same time. 
I hope that it was properly disclosed and communicated that the ZR300 was developed and tested to run on Linux platforms. librealsense is trying to provide the capabilities to get the maximum out of the device on as many platform as possible, but some problems can't be addressed without proper hardware debug. 
If you find some suitable work-around, please update this thread to help others who want to use the camera this way.

closed 2017-07-27 by oordan