realsenseai/librealsense #524: "No device detected" on Ubuntu with Intel Joule.

samcaulfield, 2017-06-07:
| Required Info  |   |
|---|---|
| Camera Model | R200 | 
| Firmware Version | 2.1.27.2853 | 
| Operating System & Version | Ubuntu 16.04.2 |
| Kernel Version (Linux Only)  | 4.8.0-54-generic |
| Build System | Makefile | 

| Platform Info | |
|---|---|
| System | Intel Joule 570X |
| BIOS Version | 1F1 |

Hi,

I am having trouble setting up the R200 with Intel Joule.

I have set up the Joule by updating the BIOS to the latest version (1F1), installing an active cooler and setting Powered Limit 1 Enable to "Enabled" and Power Limit 1 Power to "8" in the BIOS according to http://www.intel.com/content/www/us/en/support/boards-and-kits/intel-joule-kits/000023508.html. That guide is for the ZR300 but I figured the steps wouldn't hurt for R200 because they are quite generic.

I installed Ubuntu 16.04.2 on the Joule and ran a full update with
_\# apt update && apt upgrade && apt dist-upgrade_

I installed the librealsense dependencies with
_\# apt install libglfw3-dev libusb-1.0-0-dev libssl-dev pkg-config_

I successfully built and installed the library using CMake 3.5.1 and GCC 5.4.0.

I installed the udev rules as described in the instructions.
_# cp config/99-realsense-libusb.rules /etc/udev/rules.d_
_# udevadm control --reload-rules && udevadm trigger_

At this point I tried several things, each on a totally fresh install:

1. I ran scripts/patch-realsense-ubuntu-xenial.sh, which failed as in Issue #499.
2. I ran a modified version of the patch script found at https://github.com/JJHu1993/librealsense-patch which reported success.
3. I tried on an unpatched kernel.

The below steps were repeated for each of the three cases above:

I rebooted the system with the R200 plugged in. The R200 was not plugged in during install.
After boot I ran cpp-capture and the output was "No device detected. Is it plugged in?"

dmesg reports that uvcvideo was loaded with no apparent problems.

I have tried with the R200 plugged directly into the Joule and plugged into a powered USB 3.0 hub.
I have also tried this on the Ubuntu Intel Joule image, with the same result.
The R200 works on both another Joule and on my Windows machine (with the SDK examples).

I can provide more information if required.

Thanks

ev-mp, 2017-06-08:
Hi, hw logs are definitely required here : please record and share 'dmesg' and 'lsusb' report when connection the camera both to USB-A and USB-C connector.

Additionally,  you don't nee to install kernel patch if you're using the [official Ubuntu image for Joule](https://developer.ubuntu.com/core/get-started/intel-joule)

samcaulfield, 2017-06-08:
I did a fresh install of the Ubuntu Intel Joule image on the Joule and performed the installation without patching the kernel.

Unfortunately I don't have a way to connect the camera to the USB-C port.

I connected the R200 into a powered USB 3.0 hub that works with the keyboard and mouse. dmesg did not report anything new when I compared the output before and after connecting.

lsusb reports the following when the R200 is connected:
(I will edit this in 1 minute with a paste from the Joule)

ev-mp, 2017-06-08:
The device is not on the list. 
 Run "dmesg -wT" while connecting the camera to the board, first via the USB hub, then directly to the USB-A port (remove the hub) and provide the results.

samcaulfield, 2017-06-09:
Using dmesg -wT, there is no additional output when I plug the R200 into the hub.
However, when I plug it directly into the board I get the following:

```
[Fri Jun  9 12:56:37 2017] usb 2-2: new SuperSpeed USB device number 4 using xhci_hcd
[Fri Jun  9 12:56:37 2017] usb 2-2: New USB device found, idVendor=8086, idProduct=0a80
[Fri Jun  9 12:56:37 2017] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Fri Jun  9 12:56:37 2017] usb 2-2: Product: Intel RealSense 3D Camera R200
[Fri Jun  9 12:56:37 2017] usb 2-2: Manufacturer: Intel Corp
[Fri Jun  9 12:56:37 2017] usb 2-2: SerialNumber: SN_2431002863
[Fri Jun  9 12:56:38 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Fri Jun  9 12:56:38 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Fri Jun  9 12:56:38 2017] uvcvideo: Unable to create debugfs 2-4 directory.
[Fri Jun  9 12:56:38 2017] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.2/input/input8
[Fri Jun  9 12:56:38 2017] uvcvideo: Found UVC 1.10 device Intel RealSense 3D Camera R200 (8086:0a80)
[Fri Jun  9 12:56:38 2017] uvcvideo: Unable to create debugfs 2-4 directory.
[Fri Jun  9 12:56:38 2017] input: Intel RealSense 3D Camera R200 as /devices/pci0000:00/0000:00:15.0/usb2/2-2/2-2:1.4/input/input9
[Fri Jun  9 12:56:38 2017] usbcore: registered new interface driver uvcvideo
[Fri Jun  9 12:56:38 2017] USB Video Class driver (1.1.1)
```

I have performed a fresh install on a second Joule, and the R200 works with no issue using the hub after the same setup of librealsense. The only difference is that the problematic Joule has an active cooler installed and the working Joule uses a heatsink only.

ev-mp, 2017-06-14:
Hi @samcaulfield , I strongly recommend not to use USB hub when connecting cameras to Joule. Joule provides two USB3 ports, so you can connect all the peripherals to the second port, leaving USB-A dedicated for camera.
The active cooler by itself seem not to be related. Instead, I'd ask you to check whether the two Joule modules are running the same BIOS version.

samcaulfield, 2017-06-16:
Could you advise me on how to check the BIOS version? In the BIOS screen under the Intel(R) Atom(TM) line, there is a long alphanumeric string. This string is the same on both Joules.

I already tried connecting the R200 directly to the board with no success. I'll try it again just to be sure but it leads me to suspect that there is an issue with the Joule board and not on your end.

samcaulfield, 2017-07-05:
Hi, I think it is reasonably safe to believe that the issue was a defect with that particular Joule. I have the R200 running well on another Joule board. Given that the Joule is going to be discontinued soon I don't see much point in diagnosing the issue further. Thanks for the debugging help.

closed 2017-07-05 by samcaulfield